phanrahan/magma #762: coroutine non manual encoding behavior question

splhack, 2020-07-03:
trying to figure out the non manual encoding behavior with this simple coroutine

    import magma as m

    @m.syntax.coroutine
    class top:
        def __init__(self):
            self.a: m.Bits[4] = 0

        def __call__(self) -> m.Bits[4]:
            while True:
                self.a = m.bits(0, 4)
                yield m.bits(0, 4)

                self.a = m.bits(1, 4)
                yield m.bits(1, 4)

    m.compile("top", top, inline=True)

I'm assuming the first return value is `m.bits(0, 4)`, and `m.bits(1, 4)` in the next clock cycle. But the converted result is this.

    @m.circuit.sequential(reset_type=m.AsyncReset)
    class top:

        def __init__(self):
            self.a: m.Bits[4] = 0
            self.yield_state: m.Bits[1] = 0

        def __call__(self) ->m.Bits[4]:
            if self.yield_state == 0:
                self.a = m.bits(1, 4)
                self.yield_state = m.bits(1, 1)
                __magma_coroutine_return_value = m.bits(1, 4)
            else:
                self.a = m.bits(0, 4)
                self.yield_state = m.bits(0, 1)
                __magma_coroutine_return_value = m.bits(0, 4)
            return __magma_coroutine_return_value

`self.yield_state` initial value is `0`, so the first return value is `m.bits(1,4)`, and `m.bits(0,4)` in the next clock cycle. the result was flipped from my assumption.

coroutine code was parsed as two yield blocks

    self.a = m.bits(0, 4) # block 0
    yield m.bits(0, 4)    #

    self.a = m.bits(1, 4) # block 1
    yield m.bits(1, 4)    #

and added `self.yield_state` assign to go next state.

    self.a = m.bits(0, 4)           # block 0
    self.yield_state = m.bits(1, 4) # to go next state, block 1
    yield m.bits(0, 4)              #

    self.a = m.bits(1, 4)           # block 1
    self.yield_state = m.bits(0, 4) # to go next state, block 0
    yield m.bits(1, 4)              #

put it into if blocks

    if self.yield_state == m.bits(0):
        self.a = m.bits(0, 4)           # block 0
        yield m.bits(0, 4)              #
        self.yield_state = m.bits(1, 4) # to go next state, block 1

    else: # if self.yield_state == m.bits(1):
        self.a = m.bits(1, 4)           # block 1
        yield m.bits(1, 4)              #
        self.yield_state = m.bits(0, 4) # to go next state, block 0

This is my expected result. But the current result was flipped.

    if self.yield_state == 0:
        self.a = m.bits(1, 4)
        self.yield_state = m.bits(1, 1)
        __magma_coroutine_return_value = m.bits(1, 4)
    else:
        self.a = m.bits(0, 4)
        self.yield_state = m.bits(0, 1)
        __magma_coroutine_return_value = m.bits(0, 4)

It works if the initial `self.yield_state` was `m.bits(1)`. However this patch breaks test_uart.py..

    diff --git a/magma/syntax/coroutine.py b/magma/syntax/coroutine.py
    index 8ff2c016..4ef43166 100644
    --- a/magma/syntax/coroutine.py
    +++ b/magma/syntax/coroutine.py
    @@ -193,7 +193,7 @@ def _coroutine(defn_env, fn):
         yield_state_width = clog2(len(yield_encoding_map))
         if not manual_encoding:
             method_name_map["__init__"].body.append(ast.parse(
    -            f"self.yield_state: m.Bits[{yield_state_width}] = 0"
    +            f"self.yield_state: m.Bits[{yield_state_width}] = {len(yield_encoding_map) - 1}"
             ).body[0])

         # create new call body

leonardt, 2020-07-08:
The semantics of the coroutine logic is that upon RESET (initial state), the coroutine is advanced to the first yield.  So in this case, the coroutine starts at the first yield.  Then, the output is produced combinationally as the next yield statement encountered.  This explains why the initial yield state is 0, and the initial output is the second yield.  It's a bit confusing, because the output is determined by what will be yielded next given that we're at the current yield.

If you want an initial state, you can place a yield before the while true loop, then I think you would get the result you expect.  Basically, the current semantics do not provide an implicit initial state, and instead use the first yield as the initial state.

This was chosen to provide explicit control over the reset/initial state semantics, otherwise every coroutine would have an implicit reset state where you can't control the output value.  I'll see if I can write a simple explanation of this for the documentation, this would be useful for helping explain the semantics.

splhack, 2020-07-09:
Thank you for the clarification, worked as expected!

    import magma as m
    import fault

    @m.coroutine()
    class top:
        def __init__(self):
            self.a = m.Register(T=m.Bits[4], init=0)()

        def __call__(self) -> m.Bits[4]:
            yield m.bits(2, 4)
            while True:
                self.a = m.bits(0, 4)
                yield m.bits(0, 4)

                self.a = m.bits(1, 4)
                yield m.bits(1, 4)

    m.compile("top", top, inline=True)

    tester = fault.Tester(top, top.CLK)
    tester.circuit.ASYNCRESET = 0
    tester.eval()
    tester.circuit.ASYNCRESET = 1
    tester.eval()
    tester.circuit.ASYNCRESET = 0
    tester.eval()
    tester.circuit.O.expect(0)
    tester.step(2)
    tester.circuit.O.expect(1)
    tester.step(2)
    tester.circuit.O.expect(0)
    tester.step(2)
    tester.circuit.O.expect(1)
    tester.step(2)
    tester.circuit.O.expect(0)

    tester.compile_and_run(target="verilator", flags=['-Wno-fatal'])

closed 2020-07-09 by splhack