phanrahan/magma #774: magma error: test_clock_tuple_compile.py

phanrahan, 2020-07-20:
Any ideas what is causing this error? I updated coreir.

```
tests/test_coreir/test_clock_tuple_compile.py ERROR:magma:Input port Foo_inst0.clocks not driven
  Def:
    Instances:
      Foo_inst0 : Foo
    Connections:
      Foo_inst0.I <=> self.I
      Foo_inst0.O <=> self.O

ERROR: {Main}.Foo_inst0.clocks_clk Is not fully connected (N)
{Main}.Foo_inst0.clocks_rst Is not connected

I AM DYING!
Assertion failed: (0), function die, file /Users/hanrahan/git/coreir/src/ir/context.cpp, line 104.
F

=================================== FAILURES ===================================
_________________________ test_auto_wire_tuple_clocks __________________________

    def test_auto_wire_tuple_clocks():
        class Intf(m.Product):
            clk = m.In(m.Clock)
            rst = m.In(m.Reset)
    
        class Foo(m.Circuit):
            io = m.IO(clocks=Intf, I=m.In(m.Bit), O=m.Out(m.Bit))
    
        class Main(m.Circuit):
            io = m.IO(clocks=Intf, I=m.In(m.Bit), O=m.Out(m.Bit))
    
            foo = Foo()
            foo.I @= io.I
            io.O @= foo.O
    
    
>       m.compile("build/test_auto_wire_tuple_clocks", Main)

tests/test_coreir/test_clock_tuple_compile.py:22: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
magma/compile.py:62: in compile
    compiler.compile()
magma/backend/coreir_compiler.py:73: in compile
    fn()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <magma.backend.coreir_compiler.CoreIRCompiler object at 0x7fa4fad29410>

    def _compile_verilog(self):
        cmd = _make_verilog_cmd(self.deps, self.basename, self.opts)
        ret = subprocess.run(cmd, shell=True).returncode
        if ret:
>           raise RuntimeError(f"CoreIR cmd '{cmd}' failed with code {ret}")
E           RuntimeError: CoreIR cmd 'coreir -l global -i /Users/hanrahan/git/magma/tests/test_coreir/build/test_auto_wire_tuple_clocks.json -o /Users/hanrahan/git/magma/tests/test_coreir/build/test_auto_wire_tuple_clocks.v' failed with code -6

magma/backend/coreir_compiler.py:87: RuntimeError
------------------------------ Captured log call -------------------------------
ERROR    magma:logging.py:87 Input port Foo_inst0.clocks not driven
=========================== short test summary info ============================
FAILED tests/test_coreir/test_clock_tuple_compile.py::test_auto_wire_tuple_clocks
!!!!!!!!!!!!!!!!!!!!!!!!!! stopping after 1 failures !!!!!!!!!!!!!!!!!!!!!!!!!!!
=================== 1 failed, 99 passed, 3 skipped in 2.64s ====================
```

leonardt, 2020-07-20:
Hmm, can you do `$ pip show magma-lang coreir fault hwtypes` to see what versions of the various packages you're running?  I have 

```
Name: magma-lang
Version: 2.0.2
...
---
Name: coreir
Version: 2.0.86
...
---
Name: fault
Version: 3.0.7
...
---
Name: hwtypes
Version: 1.3.0
...
```

leonardt, 2020-07-20:
Also, it looks like you're using a local installation of coreir (based on /Users/hanrahan/git/coreir/src/ir/context.cpp).  One sanity check would be to uninstall this version (remove it from your path or use make uninstall in your build directory), install the latest pycoreir package (pip install --upgrade core), then re-run the test.  If this works, that would suggest a problem with your local build that we can investigate

phanrahan, 2020-07-20:
Hmm. I have the development versions checked out. They don't work?
```
% pip show magma-lang coreir fault hwtypes
Name: magma-lang
Version: 2.0.68
Summary: An embedded DSL for constructing hardware circuits
Home-page: https://github.com/phanrahan/magma
Author: None
Author-email: None
License: MIT
Location: /Users/hanrahan/miniconda3/lib/python3.7/site-packages
Requires: hwtypes, numpy, coreir, ast-tools, six, kratos, graphviz, mako, staticfg, astor, pyverilog, colorlog, dataclasses
Required-by: gemstone, canal, fault, mantle
---
Name: coreir
Version: 2.0.87
Summary: Python bindings for CoreIR
Home-page: https://github.com/leonardt/pycoreir
Author: Leonard Truong
Author-email: lenny@cs.stanford.edu
License: BSD License
Location: /Users/hanrahan/git/pycoreir
Requires: hwtypes
Required-by: magma-lang, metamapper, CoSA, fault
---
Name: fault
Version: 3.0.23
Summary: A Python package for testing hardware (part of the magma ecosystem)
Home-page: https://github.com/leonardt/fault
Author: Leonard Truong
Author-email: lenny@cs.stanford.edu
License: BSD License
Location: /Users/hanrahan/git/fault
Requires: astor, coreir, cosa, z3-solver, hwtypes, magma-lang, pyyaml, scipy, numpy, DeCiDa
Required-by: gemstone, mantle
---
Name: hwtypes
Version: 1.4.2
Summary: Python implementations of fixed size hardware types (Bit, BitVector, UInt, SInt, ...) based on the SMT-LIB2 semantics
Home-page: https://github.com/leonardt/hwtypes
Author: Leonard Truong
Author-email: lenny@cs.stanford.edu
License: UNKNOWN
Location: /Users/hanrahan/miniconda3/lib/python3.7/site-packages
Requires: gmpy2, pysmt, z3-solver
Required-by: magma-lang, gemstone, canal, CoSA, coreir, fault
```

leonardt, 2020-07-20:
The development versions should work, but I wanted to just verify that they are being used.  Can you try using the release version of the pycoreir package? Perhaps there's some problem with your local installation.

Also, can you try doing `/usr/local/bin/coreir --version` to see which version you have installed? (You may need to change the path to where you have it installed if it's in a different location).  The current master branch should be installing v0.1.40

phanrahan, 2020-07-20:
When I install coreir from pycoreir it goes into my anaconda directory. Is that ok? 

It reports v0.1.40

leonardt, 2020-07-20:
yes that's okay, does the issue persist with the pycoreir only setup?

leonardt, 2020-07-20:
Also maybe a `which -a coreir`, my suspicion is there is an old version of coreir lying around somewhere in your PATH or DYLD_LIBRARY_PATH

phanrahan, 2020-07-20:
```
% which -a coreir
/Users/hanrahan/miniconda3/bin/coreir
```

A little further:
```
=================================== FAILURES ===================================
____________________________ test_sequential2_reset ____________________________

    def test_sequential2_reset():
>       @m.sequential2(reset_type=m.AsyncReset, has_enable=True)
        class Test2:

tests/test_syntax/test_sequential2.py:372: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
magma/syntax/sequential2.py:327: in seq_inner
    file_name=file_name)(cls.__init__)
../ast_tools/ast_tools/passes/util.py:208: in __call__
    return self.exec(etree, stree, env)
../ast_tools/ast_tools/passes/util.py:218: in exec
    return exec_def_in_file(etree, env, self.path, self.file_name, stree)
../ast_tools/ast_tools/common.py:40: in exec_def_in_file
    return exec_in_file(tree, st, path, file_name, serialized_tree)[tree.name]
../ast_tools/ast_tools/common.py:58: in exec_in_file
    source = astor.to_source(tree)
../../miniconda3/lib/python3.7/site-packages/astor/code_gen.py:52: in to_source
    generator.visit(node)
../../miniconda3/lib/python3.7/site-packages/astor/node_util.py:143: in visit
    return visitor(node)
../../miniconda3/lib/python3.7/site-packages/astor/code_gen.py:320: in visit_FunctionDef
    self.body(node.body)
../../miniconda3/lib/python3.7/site-packages/astor/code_gen.py:218: in body
    self.write(*statements)
../../miniconda3/lib/python3.7/site-packages/astor/code_gen.py:168: in write
    visit(item)
../../miniconda3/lib/python3.7/site-packages/astor/node_util.py:143: in visit
    return visitor(node)
../../miniconda3/lib/python3.7/site-packages/astor/code_gen.py:278: in visit_Assign
    self.visit(node.value)
../../miniconda3/lib/python3.7/site-packages/astor/node_util.py:143: in visit
    return visitor(node)
../../miniconda3/lib/python3.7/site-packages/astor/code_gen.py:514: in visit_Call
    self.visit(node.func)
../../miniconda3/lib/python3.7/site-packages/astor/node_util.py:143: in visit
    return visitor(node)
../../miniconda3/lib/python3.7/site-packages/astor/code_gen.py:524: in visit_Call
    write(write_comma, arg, '=' if arg else '**', keyword.value)
../../miniconda3/lib/python3.7/site-packages/astor/code_gen.py:168: in write
    visit(item)
../../miniconda3/lib/python3.7/site-packages/astor/node_util.py:143: in visit
    return visitor(node)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

node = <_ast.Constant object at 0x7fd9de825ad0>

    def abort_visit(node):  # XXX: self?
        msg = 'No defined handler for node of type %s'
>       raise AttributeError(msg % node.__class__.__name__)
E       AttributeError: No defined handler for node of type Constant

../../miniconda3/lib/python3.7/site-packages/astor/node_util.py:137: AttributeError
__________________________________ test_jtag ___________________________________

    def test_jtag():
        TEST_LOGIC_RESET = m.bits(15, 4)
        RUN_TEST_IDLE = m.bits(12, 4)
        SELECT_DR_SCAN = m.bits(7, 4)
        CAPTURE_DR = m.bits(6, 4)
        SHIFT_DR = m.bits(2, 4)
[... 241 lines cut ...]
    return visitor(node)
../../miniconda3/lib/python3.7/site-packages/astor/code_gen.py:524: in visit_Call
    write(write_comma, arg, '=' if arg else '**', keyword.value)
../../miniconda3/lib/python3.7/site-packages/astor/code_gen.py:168: in write
    visit(item)
../../miniconda3/lib/python3.7/site-packages/astor/node_util.py:143: in visit
    return visitor(node)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

node = <_ast.Constant object at 0x7fd9dede0e50>

    def abort_visit(node):  # XXX: self?
        msg = 'No defined handler for node of type %s'
>       raise AttributeError(msg % node.__class__.__name__)
E       AttributeError: No defined handler for node of type Constant

../../miniconda3/lib/python3.7/site-packages/astor/node_util.py:137: AttributeError
=========================== short test summary info ============================
FAILED tests/test_syntax/test_sequential2.py::test_sequential2_reset - Attrib...
FAILED tests/test_syntax/test_coroutine/test_jtag.py::test_jtag - AttributeEr...
FAILED tests/test_syntax/test_coroutine/test_sdram.py::test_sdram - Attribute...
FAILED tests/test_syntax/test_coroutine/test_uart.py::test_uart - AttributeEr...
============= 4 failed, 611 passed, 6 skipped in 371.16s (0:06:11) =============
% 
```

leonardt, 2020-07-20:
Can you try `pip install --upgrade astor`? It looks like support for Constant was added in a recent version (see https://github.com/berkerpeksag/astor/commit/641390339bea88f13bcdd0abaa576e078191a86c)

phanrahan, 2020-07-20:
Yeah. That did it. The tests pass now.

closed 2020-07-20 by phanrahan