phanrahan/magma #781: inline_combination without return value?

splhack, 2020-07-25:
Is this code valid as an inline combinational code?
(just deleted `O1 = reg.O` lines from tests/test_syntax/test_inline_comb.py)

    @m.inline_combinational()
    def logic():
        if io.invert:
            reg.I @= ~reg.O
        else:
            reg.I @= reg.O

If so, how about this? using `m.wire` instead of `@=`.

    @m.inline_combinational()
    def logic():
        if io.invert:
            m.wire(~reg.O, reg.I)
        else:
            m.wire(reg.O, reg.I)

both are same code but the latter causes error, because the logic function doesn't return value.

    >       retval = ast.Tuple(elts, ast.Load()) if len(elts) > 1 else elts[0]
    E       IndexError: list index out of range

leonardt, 2020-07-26:
hmm those should work, looks like a bug

leonardt, 2020-07-27:
Ah, I see the first one works as expected, the second one fails for the reason fixed in your PR (right now inline_comb assumes you use @= rather than wire).

PR #782 (2020-07-28): inline_combinational: support m.wire
Fix #781
closed 2020-07-28 by leonardt