phanrahan/magma #790: test_old_io_syntax_define.py fails

phanrahan, 2020-08-04:
Not sure why this test is failing, I think I have updated everything.

```
tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py .F [ 20%]

=================================== FAILURES ===================================
_________________________ test_simple_def[coreir-json] _________________________

target = 'coreir', suffix = 'json'

    @pytest.mark.parametrize("target,suffix",
                             [("verilog", "v"), ("coreir", "json")])
    def test_simple_def(target, suffix):
        m.config.set_debug_mode(True)
        m.set_codegen_debug_info(True)
        with pytest.warns(DeprecationWarning):
            And2 = m.DeclareCircuit('And2', "I0", m.In(m.Bit), "I1", m.In(m.Bit),
                                    "O", m.Out(m.Bit))
    
            main = m.DefineCircuit("main", "I", m.In(m.Bits[2]), "O", m.Out(m.Bit))
    
            and2 = And2()
    
            m.wire(main.I[0], and2.I0)
            m.wire(main.I[1], and2.I1)
            m.wire(and2.O, main.O)
    
            m.EndCircuit()
    
        m.compile("build/test_simple_def", main, output=target)
>       assert check_files_equal(__file__, f"build/test_simple_def.{suffix}",
                                 f"gold/test_simple_def.{suffix}")
E       AssertionError: assert False
E        +  where False = check_files_equal('/Users/hanrahan/git/magma/tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py', 'build/test_simple_def.json', 'gold/test_simple_def.json')

tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py:28: AssertionError
----------------------------- Captured stdout call -----------------------------
/Users/hanrahan/git/magma/tests/test_deprecated/test_old_io_syntax/build/test_simple_def.json
/Users/hanrahan/git/magma/tests/test_deprecated/test_old_io_syntax/build/test_simple_def.json /Users/hanrahan/git/magma/tests/test_deprecated/test_old_io_syntax/gold/test_simple_def.json
----------------------------- Captured stderr call -----------------------------
WARNING:magma:tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py:15: 'IO = [...]' syntax is deprecated, use 'io = IO(...)' syntax instead
>>                                 "O", m.Out(m.Bit))
WARNING:magma:tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py:17: 'IO = [...]' syntax is deprecated, use 'io = IO(...)' syntax instead
>>         main = m.DefineCircuit("main", "I", m.In(m.Bits[2]), "O", m.Out(m.Bit))
--- gold/test_simple_def.json
+++ build/test_simple_def.json
@@ -8,7 +8,7 @@
           ["I1","BitIn"],
           ["O","Bit"]
         ]],
-        "metadata":{"filename":"tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py","lineno":"14"}
+        "metadata":{"filename":"tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py","lineno":"15"}
       },
       "main":{
         "type":["Record",[
------------------------------ Captured log call -------------------------------
WARNING  magma:logging.py:87 tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py:15: 'IO = [...]' syntax is deprecated, use 'io = IO(...)' syntax instead
>>                                 "O", m.Out(m.Bit))
WARNING  magma:logging.py:87 tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py:17: 'IO = [...]' syntax is deprecated, use 'io = IO(...)' syntax instead
>>         main = m.DefineCircuit("main", "I", m.In(m.Bits[2]), "O", m.Out(m.Bit))
=========================== short test summary info ============================
FAILED tests/test_deprecated/test_old_io_syntax/test_old_io_syntax_define.py::test_simple_def[coreir-json]
!!!!!!!!!!!!!!!!!!!!!!!!!! stopping after 1 failures !!!!!!!!!!!!!!!!!!!!!!!!!!!
=================== 1 failed, 122 passed, 3 skipped in 9.43s ===================
% 
```

rsetaluri, 2020-08-04:
Taking a look...

leonardt, 2020-08-04:
This is a python 3.7/3.8 issue, see https://github.com/phanrahan/magma/issues/704

We could update the test to work with python 3.7, but if you're open to upgrading your python to 3.8 I think that would be simpler

rsetaluri, 2020-08-07:
@phanrahan did python 3.8 fix this?