phanrahan/magma #794: efficient HW using hwtypes for get/set_slice

kavyasreedhar, 2020-08-07:
The magma get/set_slice implementation from Lenny is at https://github.com/phanrahan/magma/blob/master/magma/primitives/slice.py, while Caleb's implementation for hwtypes is at https://github.com/StanfordAHA/lake/blob/accessor/lake/modules/hwtypes/utils.py.

Like we discussed, these implementations are fairly different and the hardware generated by the hwtypes functions isn't so great in terms of area and is not using the magma get/set_slice functions. Would anyone here be able to help with translating Lenny's functions into hwtypes utils so I could try using those instead? Thanks!

leonardt, 2020-08-10:
I think the main difference is the magma implementation's use muxes instead of shift/mask ops.  The get_slice operator basically takes a bit vector and produces a 2-d bit vector that contains the possible windows of size width, and uses the start index value to index into the 2-d vector.  

The set_slice operator produces a bit_vector where each bit that could be set has a mux that checks whether that index is in the slice range or not.  If it's in the slice range, use the corresponding bit in input value (computed dynamically based on the start value from the static index value).

kavyasreedhar, 2020-08-11:
I think that makes sense. Would Caleb be the best person to ask about translating them (in which case, I'll slack him after the asplos deadline)?