phanrahan/magma #805: ModuleNotFoundError: No module named 'magma.smart' while 'import magma'

Ntimesp, 2020-08-20:
Since I do not have python-3.7 on my computer I used an official python docker image.

here is my Dockerfile
```
From python:3.7.2
 
WORKDIR /root
 
RUN apt-get -y update && apt-get install -y libgmp-dev libmpfr-dev libmpc-dev
RUN pip install fault magma-lang mantle
```

I build this image and try to use it
```
docker build -t pytest .
docker run -it --name test --rm pytest python
```

then I get the following error
```
Python 3.7.2 (default, Mar 27 2019, 08:41:08) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import magma
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/site-packages/magma/__init__.py", line 78, in <module>
    import magma.smart
ModuleNotFoundError: No module named 'magma.smart'
>>> 
```

rsetaluri, 2020-08-20:
Looks like something was broken recently. I will release a fix shortly. Can you see what version of magma was pulled? You can try `pip freeze | grep magma`. Thanks!

Ntimesp, 2020-08-20:
magma-lang==2.1.0

Ntimesp, 2020-08-20:
I rebuild the docker image and it looks ok now.

closed 2020-08-20 by Ntimesp

rsetaluri, 2020-08-20:
Thanks for confirming. I released a fix on the latest version.