phanrahan/magma #812: Altering locals() not guaranteed to work

leonardt, 2020-08-31:
Amazingly, we've been able to alter the locals() dictionary without problem, but I recently encountered a problem where it wasn't working.  After some browsing I found that it's actually documented that changing the value of locals() is not guaranteed to work: https://docs.python.org/3/library/functions.html#locals (cc @cdonovick), so even passing it from the user scope won't save us here.

> Note The contents of this dictionary should not be modified; changes may not affect the values of local and free variables used by the interpreter.

We'll need to find a different pattern for supporting our syntaxes that don't involve modifications to locals.

cdonovick, 2020-08-31:
Im not sure how you are editing locals that isn't supported.  

Basically the following probably wont work:
```Python
locals()['x'] = 5
assert x == 5
```

However modifying the dict return from locals and execing should always works e.g.:
```Python
d = locals()
d['x'] = 5
exec('assert x == 5', d)
```

leonardt, 2020-08-31:
hmm, I think it's the pattern of modifying the enclosing scope, so perhaps that's not an issue for you flows.  inline_combinational was editing the calling scope (passed a locals() dict to some code, introduced new assignments, use new assignments in the enclosing scope).  So while modifying the dictionary and using it will work, it won't reliably work for altering an enclosing scope.

cdonovick, 2020-08-31:
Makes sense, to my knowledge there is no way to inject names into a scope without execing.

leonardt, 2020-08-31:
It turns out, in this case, I am doing `exec("x = 3", _globals, _locals)` where _locals is coming from the enclosing scope, but that still won't edit the enclosing scope so that's not reliable.  Here's a simple instance of this problem: https://stackoverflow.com/questions/40638179/nameerror-for-variable-assigned-in-exec-although-it-exists-in-locals?noredirect=1&lq=1

cdonovick, 2020-08-31:
yeah I wouldn't expect that to work.  

I don't know the full details of how in comb works maybe we should meet to brainstorm?

leonardt, 2020-08-31:
Here's an overview.  I think one way to avoid the issue is to do an AST rewrite at the class level, which would avoid the need for doing any enclosing scope magic.  The downside to this approach is I'm not sure if there's a way to do this without requiring a decorator on the class (which is annoying because ideally we'd just have to subclass circuit without extra decorators for other syntax, but not that big a deal).  The other problem is that even if we do the decorator+rewrite approach, there's the issue with the fact that the class code will run before we consume it in the decorator (where we then would rewrite and re-exec it).  I'm not exactly sure if this will be problematic, but it will limit our syntax choices and side effects are likely to be problematic to deal with.

leonardt, 2020-08-31:
Hmm, another option is to require the user only refer to defined values inside the combinational block, which would avoid the need for the variables to escape the scope of the function.  This might actually be a good thing because it doesn't introduce any magic semantics into Python.  Wiring to things in the outer scope still works because we just use object references (no need to assign into an environment).  If the user wants to store a "temporary variable", they could define a temporary magma value (e.g. `my_temp = Bits[5](name="my_temp")`) and wire it up inside the combinational block.

leonardt, 2020-08-31:
For clarification, the issue arises when we assign a variable inside the combinational block that we'd like to refer to afterwards (https://github.com/phanrahan/magma/blob/1c2f42e7de485b8ee5ecc57408ff41f45d460d98/tests/test_syntax/test_inline_comb.py#L62).

A similar approach would be to require the user to return any values that they would like to use and require that the user call the inline_comb function to retrieve them.  This adds some ergonomic overhead because they have to return the values explicitly and assign them again, so not ideal.

cdonovick, 2020-08-31:
I feel like this could be solved easily if this bug was addressed: https://bugs.python.org/issue17421

leonardt, 2020-08-31:
That would be quite nice, is there a problem with subclassing dict though? (Rather than using MutableMapping)

cdonovick, 2020-08-31:
Subclassing dict is weird and sometimes breaks https://treyhunner.com/2019/04/why-you-shouldnt-inherit-from-list-and-dict-in-python/

You could try it?

cdonovick, 2020-08-31:
In addition to no mixins you can't inherit most builtins and other types.

PR #843 (2020-09-15): Restrict inline_combinational to use wiring
Avoids the problem discussed in #812

Since we cannot guarantee that the assignment values can be used in the
outer scope, we restrict that the user use wiring.  If the user wants a
value to "escape" to the outer scope, they can create a temporary value
and wire up to it.

Also fixes a wiring bug in the inline comb logic (it should be wire
driver to drivee, this matters for using temporary values)

Fixes #812
commit 2a426038 (2020-09-15): Restrict inline_combinational to use wiring

Avoids the problem discussed in https://github.com/phanrahan/magma/issues/812

Since we cannot guarantee that the assignment values can be used in the
outer scope, we restrict that the user use wiring.  If the user wants a
value to "escape" to the outer scope, they can create a temporary value
and wire up to it.

Also fixes a wiring bug in the inline comb logic (it should be wire
driver to drivee, this matters for using temporary values)

Fixes #812
closed 2020-09-15 by leonardt