phanrahan/magma #845: Products, Structural Type Checking, and Type Equality

leonardt, 2020-09-16:
This came up when fixing an issue in #784 (specifically this commit https://github.com/phanrahan/magma/pull/784/commits/c02c691e36c743a4fe55dd988bf0cd5dc2f3c9a7)

Basically, when using `m.namedtuple` this creates a "new" product type instance from the named tuple that doesn't match the user defined product type for a port (since type equality use either "is" or subclassing logic).  In this case, I think we want to use structural type checking, where it doesn't matter if the types are the same ("is") or subclasses, but instead that they have the same structure (same fields for a product, same N/T for an array).  This problem doesn't really come up for Arrays since we use cacheing, but it can come up for Products when different names are used since m.namedtuple just uses a generic name "anon" inside named tuple (it's cached based on the name rather than structure).  Another option is to cache types based on structure, but this would prevent the use of strict equality (rather than structural equality) if needed.

Related is https://github.com/phanrahan/magma/issues/459 which asks about "type equality".  I think we've mostly been using "is" for type equality since we cache types.  One thought is to use "==" for structural type checking.  I think in the magma context, "==" returning true could mean that two types can be wired together (this is a bit more than just checking structure, since we would also check directionality).  If we can enforce the cacheing invariant, then this gives us "is" for nominal type equality and "==" for structural type equality.

Another approach to avoiding this issue is to drop m.namedtuple as a generic product constructor and require the user explicitly invoke the right type, e.g. "MyProduct(a=x, b=y)".  Then we could go back to using nominal equality since m.namedtuple doesn't need to generate a generic type.  

Any thoughts?

Magma was originally designed to use structural typing for wiring, see https://github.com/phanrahan/magma/issues/457.  I think perhaps the current code isn't faithful to this goal so it's a good opportunity to fix this up and revisit the design if desired.

I think it might be reasonable to have structural typing be the core, and allow the user to require stricter (e.g. nominal) type checking if desired.  For example, using this pattern we could have Bit be a standard structural type, and Clock inherit from Bit but then restrict wiring to be nominal (with other Clocks or subclasses of Clock).  This approach maps nicely to "is" vs "==" (we can use is for nominal type checking and "==" for structural type checking).  It also maps nicely to Python's duck typing (which is related to structural typing, but only cares about the subset of a type that's being used).  

Here's a case for structural typing: multiple libraries can all define their own AXI type (tuple containing ports).  These libraries can be composed with wiring since their structures match.  If we used nominal typing instead, these libraries would need to use a common shared AXI type, or some form of explicit conversion/cast would be necessary to compose these types (and presumably the conversion/cast would essentially be enforcing structural equality).  

One case for nominal typing would be to enforce self contained abstractions (e.g. a user can't define their own type that works with a set of abstractions).  This seems useful for encapsulation patterns for internal code, where certain components are only meant to be used in a single code base and very specific requirements on how they are used can be enforced.

Looking at these two cases, it seems that structural seems like a solid choice for the default as it facilitates composition and interoperability, and nominal is useful for user code with very specific use cases.  Are there any other cases for structural vs nominal as the default type checking behavior?

A related syntactic issue is "==" versus the notion of wireability.  We could either use "==" for structural equality (ignoring direction) and another interface (i.e. method) for checking whether two types can be wired (structurally equal and directions match).  Or we could just use "==" to mean two types can be wired together (or structural equality incorporates directionality).  This actually comes up quite a bit in the code base, where we want to check whether two types match and can be wired together.

rsetaluri, 2020-09-16:
> I think it might be reasonable to have structural typing be the core, and allow the user to require stricter (e.g. nominal) type checking if desired. For example, using this pattern we could have [...]

I think this is the ideal path forward. Not sure what the code/implementation challenges are so that may ultimately influence what we do (we ideally don't want to have to do a large rewrite).

> A related syntactic issue is "==" versus the notion of wireability. We could either use "==" for structural equality (ignoring direction) and another interface (i.e. method) for checking whether two [...]

I'm not a huge fan of overloading `==` to mean "wireable". It seems just as easy to require (or allow with a default impl) types to provide a `wireable` method. Then `is` means the python `is`, and `wireable()` means magma wireable.

leonardt, 2020-09-16:
So `==` will mean structural equality, e.g.
* In(Bit) == Out(Bit)
* Array[2, In(Bit)] == Array[2, Out(Bit)]
* Tuple[In(Bit), Out(Bit)] == Tuple[Out(Bit), In(Bit)]

Then wireable could first check `==`, then check directions of the leaf nodes (seems like we could abstract this if recursive types provide a "children" method)

leonardt, 2020-09-16:
I don't think there's too much code that depends on `==` right now since it's not really well defined, most of the code use `is` or `issubclass`.  I think we could add this "wireable" interface and start using it in places where issubclass is used (e.g. when checking the types of mux inputs, we could infer a type T and then check that all values are of a type wireable to T).  With this in mind, we could define an internal method `wireable(x, y)` to help with implicit coercion (e.g. if x or y is a coercible type like int/bool, we call the constructor of the other value to do the coercion)

rsetaluri, 2020-09-17:
That all sounds pretty good to me. Using `==` to mean "structural equality" seems reasonable as well.

closed 2020-09-22 by leonardt