phanrahan/magma #851: [RFC] Conditional Wiring

leonardt, 2020-09-21:
**Requirement:** Provide a language construct for conditional Wiring.  This draws from the design of Chisel which uses "when" for conditional assignment, a convenient construct for defining control behavior.

**Design Goals**
* Use different syntax than `if` so that designers can clearly distinguish between hardware behavior and Python metaprogram
* Support arbitrary nesting of code blocks (should not restrict the user from interleaving Python `if` statements or other useful metaprogramming constructs)
* Do not require AST rewrites for easier error messages and debugging

**Proposed Syntax**
We can draw off chipy's use of the with statement (see https://github.com/chipy-hdl/chipy#a-simple-example).  This can be done by maintaining a stack of active conditions (scoped per circuit definition).  When wiring with an active condition stack, the driver is staged with the conditions stored.  Multiple with statements can be used to capture elif and else patterns.  At the end of circuit definition, the logic to generate the conditional assignments is processed ensuring that the elif/else priority is captured (we delay actually doing this wiring since the user can build up to a default else at the end, so we don't want to start wiring until we have the else).  

Here's a possible syntax:
```python
class Foo(m.Circuit):
    io = m.IO(I0=m.In(m.Bits[4]), I1=m.In(m.Bits[4]), S=m.In(m.Bits[2]), O=m.Out(m.Bits[4]))
    with m.When(io.S == 0):
        io.O @= io.I0
    with m.ElseWhen(io.S == 1):
        io.O @= io.I1
    with m.Else:
        io.O @= 7
```

**Notes**
* Using the `with` statement and wrapping conditions inside a magma construct (e.g. `m.When`) allows us to use the context manager pattern to easily manage a condition stack.  `with` as a statement allows us to nest arbitrary code inside the block.
* One thing to consider is enforcing valid syntax.  For example, the user could do something like:
```python
with m.When(x):
     ...
with m.Else:
     ...
with m.ElseWhen(z):
     ...
```
   In this case, we could have `Else` clear the current When chain in the stack, and the ElseWhen could raise an Error that there is not active When chain (need to call When first).  We could do something similar for Else without a When.  Are there any other cases of "invalid" syntax that we need to beware of?  In theory the user could put random code between components of a chain, this wouldn't be necessarily a problem but could make code hard to read.  This could easily be caught by a lint-style analysis tool if we wanted.

rsetaluri, 2020-09-21:
This seems reasonable overall. However, it's not immediately clear to me how the bodies of the `with`'s can be captured and staged. Briefly, do you have an idea of how to implement that?

(I get the idea of maintaining a stack etc. in general, but I'm not sure how to capture all the magma code inside the blocks and execute them with magma-level predication -- without doing something like ast parsing/exec'ing/etc.)

leonardt, 2020-09-21:
Basically inside the implementation of wire (which @= uses) we check if there's any active conditions on the stack.  If so, we add them to the list of conditional drivers, otherwise, we do the normal wiring logic.  

I'll have to think some more, but I think just extending the behavior of wiring to capture this is enough.  Instances can just be done as normal since they'll need to exist regardless, so we just need a hook before wiring to check whether the current wiring logic is conditional and if so, ensure that the conditional logic is generated.

rsetaluri, 2020-09-21:
Conceptually that makes sense. My only worry would be if intervening in wiring would be too intrusive and not robust. E.g. what stuff (e.g. outside of wiring) would not get the right treatment, how hard is it to maintain [e.g. new type with custom wire call has to implement this?], etc.

...but I'm not sure I see any other (good) way to do it right now. If we wanted to accept inspecting ast/exec'ing, I think there would be another way to do it, but that of course has a slew of issues as well. I do think just as we've had a design doc of options for syntax, it would be good to do the same for implementations -- if there are multiple options at all.

leonardt, 2020-09-21:
Assuming the new type defining a custom wire just recursively dispatches to the core magma wiring logic, it should work fine (the new type just defines wiring semantics in terms of core magma wiring, and core magma defines conditional wire semantics)

leonardt, 2020-09-21:
I think this only relates to wiring (so @= behavior is changed when inside a with statement), so conceptually I don't think this should affect any other parts of the core (we'll see about this in practice, but that's the design goal)

rsetaluri, 2020-09-21:
Regarding the "validity" of syntax, there might be an easier (but "less elegant") syntax to explicitly relate if-elif-else blocks. Imagine something like

```python
cond = m.Conditional()
with cond.when(x):
     ...
with cond.elif(y):
     ...
with cond.else(z):
    ...
with cond.elif(w):  # errors!!
    ...
```

if you want to avoid an extra line:
```python
with m.When(x) as cond:
    ...
with cond.else(y):
    ....
```

something to this effect could help relate the various blocks more directly. it may not be necessary, but at the same time it might very well be possible that it's impossible not to create an ambiguous scenario that has to be explicitly disambiguated or we have to make an assumption somewhere.

rsetaluri, 2020-09-21:
> Assuming the new type defining a custom wire just recursively dispatches to the core magma wiring logic, it should work fine (the new type just defines wiring semantics in terms of core magma wiring, and core magma defines conditional wire semantics)
> 
> I think this only relates to wiring (so @= behavior is changed when inside a with statement), so conceptually I don't think this should affect any other parts of the core (we'll see about this in practice, but that's the design goal)

This all makes sense. It still feels unsavory to me that something as fundamental as wiring (perhaps the lowest level abstraction in magma) knows about a high-level ("behavioral"!!) construct such as wiring. It might be in the class "let's implement it and see".

Would also be interested in hearing @cdonovick 's thoughts.

cdonovick, 2020-09-21:
First, I am unconvinced that "Do not require AST rewrites" necessarily implies  "easier error messages and debugging" when things get this magical.

How is the condition stack closed?  Will it close automatically for me at the end of circuit? What happens if I don't have an else? What if my cases are exhaustive? 

I don't know how circuit definition scoping work at this point so this might be an issue but what happens if I leave it open and then define a new a circuit before the end of a circuit definition?

```Python
def generate(recurse=False):
    class Foo(m.Circuit):
        io = m.IO(I0=m.In(m.Bits[4]), I1=m.In(m.Bits[4]), S=m.In(m.Bits[2]), O=m.Out(m.Bits[4]))
        with m.When(io.S == 0):
            io.O @= io.I0

        if recurse:
            # okay this pretty stupid but demonstrates my question
            generate()
            
        with m.Else:
            io.O @= 7

    return Foo
Foo = generate(True)
```

Can I have multiple whens open at a time?  What happens if they conflict in some way?

```Python
        with m.When(io.S1 == 0):
            io.O @= io.I0

        with m.When(io.S2 == 0):
            io.O @= io.I1
```

What's the advantage of this over some syntax which is less magical like?

```Python
io.O @= Case(
   (io.S == 0, io.I0),
   (io.S == 1, io.I1),
   (True, 7)
)
```

rsetaluri, 2020-09-21:
Also would want to make sure weird overlapping cases like

```python
when x:
    a[:10] @= foo
else:
    a[5:15] @= bar
```

work

leonardt, 2020-09-22:
> First, I am unconvinced that "Do not require AST rewrites" necessarily implies "easier error messages and debugging" when things get this magical.

In this syntax, if we get a wiring error, e.g. the user write `x @= 3` and x is a tuple, we can simply raise an exception and it will traceback to the source line using built-in Python.  It's harder to get easier than free when it comes to error messages like that (compared to having an exception raised that traces back to a generated string/file being execed).

The condition stack closes when:
1. an else is encountered
2. a new when is encountered
3. a circuit definition is closed

2. presents a case for confusion that we have discussed above and I think Raj has some decent syntax options for improving clarity.  Basically, a user could leave an elif chain open, write some arbitrary code, then add another elif that would continue along the open condition chain, however I don't know if this is really something we need to worry too much about, since it's not really a natural pattern (would you really imagine yourself starting with an elsewhere and appending to the current chain on accident? Or is it pretty obvious that you start with a When, which would close the previous chain)

Leaving a condition stack open and starting a new circuit is actually a desired feature.  The code you showed is precisely what we want to support, interleaved conditional logic and Python meta programming.  Magma already has a notion of a definition stack, so we can associate open condition stacks with a specific definition, so scoping is not an issue (i.e. we already have a notion of definition scoping).  TLDR, your example would work just fine.

> Can I have multiple whens open at a time? What happens if they conflict in some way?
Opening a new when will close the previous when.  This is the main problem with this syntax in my opinion.  The chisel version forces you to keep when/elsewhen/else blocks together because they use method chaining.  The python with statement doesn't give us any syntactic mechanism to enforce this.  Ideally your example would be like having two if statements in combinational.  Chisel uses last connect semantics for this which I think seems pretty reasonable.  Basically if both conditions fire, the code in the later when statement has priority.

> What's the advantage of this over some syntax which is less magical like?
The Case syntax doesn't let you nest arbitrary python code in the body (e.g. the if statement in your previous example), but would be useful and a decent option for implementing Case statements (especially if we could carry them through CoreIR to verilog).  One nice feature of when as a statement is that we can collect multiple conditional wires inside the same statement (compared to having to generate separate Case statements for each wire).

For raj's weird overlapping case, I think that should work fine, assuming the `a` has a default driver (or else a[0:5] would be unconnected when not x and a[10:15] when x).  Again, here we would use the last connect semantics, so if different when blocks touched different overlapping parts of a value, they would need need to ensure priority is given to the later statement (in this case, we'd likely just recurse the logic to the underlying child elements which would use the standard priority semantics).

As someone who has plenty of experience working with AST rewriting and dealing with the issues it brings along, I really do feel like there's a lot to gain if we can massage this syntax into standard Python.  I think the implementation involves three things:
1. Maintain a global stack of conditions
2. Update wiring to be aware of the condition stack and keep track of conditional wires
3. Finalize circuits by generating the logic to implement the conditional wiring

Steps 2/3 could be swapped for generating the logic on the fly as conditional wires are added, we do this in https://github.com/phanrahan/magma/pull/850 for the ready/valid enq/deq methods.  I'm not sure if there's any strong argument for either strategy, but I think staging it is probably cleaner than continuously modifying (unwiring/wiring as assignments are added).

leonardt, 2020-09-22:
Also re: wiring be aware of this behavioral construct, it's worth noting that we could constrain this to the implementation only.  That is, it's convenient to have wiring aware of this so that we can avoid rewriting the AST and get error messages for free.

In the language specification, we could have this defined purely as syntax sugar for rewriting the code to the equivalent muxes, so the core language remains built around instances and wiring (no awareness of conditionals).  So with this view, we could regard it as a language extension that compiles to the core language, which could be useful for things like formal verification of these semantics or formal reasoning about the language, but I think implementation wise we want to place heavy consideration on the user experience.

rsetaluri, 2020-09-22:
That all makes sense. The one thing I want to make sure is still allowed is doing ad-hoc (i.e. *outside* of a circuit definition) stuff such as `m.Bit() @= m.Bit()`. If wiring checks for when-stack etc., I want to make sure that in the case that the previous code is run context-free, it still works.

I think my concerns are all addressed at this point. Think we can work out issues in the implementation as we get it.

cdonovick, 2020-09-22:
> I don't know if this is really something we need to worry too much about, since it's not really a natural pattern (would you really imagine yourself starting with an elsewhere and appending to the [...]

This is precisely my concern.  Have you ever refactored your code and left an if statement starting with an `elif`?  Now imagine that wasn't a syntax error but instead it just got lumped onto whatever the last `if` you used. Talk about hard to debug...

> Opening a new when will close the previous when

I would rather opening a new `when` raise an error if the previous when wasn't explicitly closed. For similar reasons to the above.  

> The Case syntax doesn't let you nest arbitrary python code in the body

Why do I need this?  What's the difference between:

```Python
x = io.I0 + 1
y = io.I1 - 1
io.O @= Case(
   (io.S == 0, x),
   (io.S == 1, y),
   (True, 7)
)
```
and
```Python
with m.When(io.S == 0):
    x = io.I0 + 1
    io.O @= x
with m.ElseWhen(io.S == 1):
    y = io.I1 - 1
    io.O @= y
with m.Else:
    io.O @= 7
```

> One nice feature of when as a statement is that we can collect multiple conditional wires inside the same statement (compared to having to generate separate Case statements for each wire

Okay that doesn't work with my case syntax but I am unconvinced that it is such a big win but I could be wrong, it's happened at least once before.  Further instead of my initial syntax of  `<TARGRET> @= Case((<GAURD>, <VALUE>)+)` it could be `Case((<GAURD>, (<TARGET>, <VALUE>)+)+)`

> As someone who has plenty of experience working with AST rewriting and dealing with the issues it brings along, I really do feel like there's a lot to gain if we can massage this syntax into standard Python. 

No disagreement, I am just concerned with how stateful this solution is -- its very magma 0.1 with the open circuit definition close circuit definition functions (or whatever the terminology was).   Its just bound to be a headache in my estimation.

cdonovick, 2020-09-22:
Idea based on raj's proposal that would address a lot of my concerns
```Python
with m.When(io.S == 0) as cond:
   ...
# __exit__ would put cond into a state so that ElseWhen cold be called on it.
# ElseWhen cannot be called prior to __exit__
cond_alias = cond
with cond.ElseWhen(...) as cond:
    ...
# __exit__ invalidates the the old cond so cond_alias could cannot be used
with cond.Else as cond:
    ...    
# __exit__ invalidates both old cond and the new cond
```
Now you could still accidentally continue an old if but if you made different names for each `cond`  it could be avoided.

This would potentially allow for nested conditions which I think would be forbidden by the initial proposal e.g.:
```Python
with m.When(io.S0 == 0) as cond:
    with m.When(io.S1 == 0) as cond2:
        io.O @= ...
    with cond2.ElseWhen(io.S1 == 1) as cond2:
        ...
```

leonardt, 2020-09-22:
Nesting would and should be supported (the original proposal could do it by keeping track of conditions that haven't been exited yet).

But I think the refactoring issue is valid, and using a cond variable seems like a reasonable way to keep track of related cases explicitly.  But yet it seems we would want to enforce some style where conds have unique names. Could be annoying for the user to have to come up with them but probably better than floating when statements.

leonardt, 2020-09-22:
You need the ability to nest inside case if you want to have a python if statement inside (which when would allow). Otherwise you have to move your if statement to the outside and have two separate case statements. It's possible to do, but not the most ergonomic option since it could possibly require duplicating quite a bit of code if you have to lift all your python code to the outside.

leonardt, 2020-09-22:
One thought, we could use an AST visitor to validate the code and raise an error if we encounter a bad pattern. So if we chose this approach, we could force when/elsewhen/else statements to be grouped together in the same format as if statements (so raise an error if we find an elsewhen without a preceding when).  Since we have access to the file/line in the AST we could probably generate a nice error message easily

cdonovick, 2020-09-22:
> Nesting would and should be supported (the original proposal could do it by keeping track of conditions that haven't been exited yet).

I thought a new when closed the previously open one?  I guess if it wasn't really open until `__exit__` that would work but seems tricky to implement, 

> You need the ability to nest inside case if you want to have a python if statement inside (which when would allow). Otherwise you have to move your if statement to the outside and have two separate [...]

I think I would need to see a use case.  Everyone I come up with can be just pushed outside the block without a problem.

> One thought, we could use an AST visitor to validate the code and raise an error if we encounter a bad pattern. So if we chose this approach, we could force when/elsewhen/else statements to be [...]

I think a linter would make me happy with this solution.  Do you imagine it being run automatically somehow?
This could actually open a number of "stateful syntaxes" in other areas as well.  Having a magma linter could 
alleviate the tension between robustness and the limits of python we often run into.  We don't want to rewrite the AST
when we don't have to and we definitely don't want to have a custom interpreter this a perfect middle ground.

leonardt, 2020-09-22:
Hmm I think you can push it out, but you have to introduce temporaries (like x, y in your example), which could be annoying for larger examples.  Another pattern would be only conditionally assigning to values in some cases.  So even though you could make your case variadic, it might get annoying with only certain cases assigning certain values (would have to specify the default value for all the other cases, rather than leaving it implicit).

Thinking about the linter: the obvious ways to do it are using a class decorator or inside the metaclass pipeline and using inspect.  The problem here is that the linter runs after the class body is executed.  This might not be a problem as long as code that would be caught by lint doesn't cause a runtime exception before hand, which I think might be the case (you might get the wrong logic generated for your conditional assign, but that's not a problem since lint will throw an error later).  In this situation, the linter would only be catching syntax level patterns that indicate possibility of logical mistakes, but should be valid code w.r.t. to triggering runtime errors.  I'm not 100% certain of this though, I'll have to think, but maybe there's a case where you get an else chain wrong and it triggers an error?

Obviously, more generally, it would be ideal if we could run the linter before the code is executed, but I'm not sure we have a hook that would allow us to do this (e.g. in theory if prepare gave us access to the class object being defined and we could fetch it's source code, we could run it there, but alas, it does not).

Another option would be to run this from the CLI (e.g. instead of `python myfile.py`, we run `magma myfile.py` and this runs a linter).  This may solve the problem of running the linter before the class code is executed, but would likely prove difficult to implement in general (e.g. how do we know what are magma circuits? What if someone just does `Foo = m.Circuit` and inherits from `Foo`?).

It seems like running it from the metaclass pipeline (since we already hook into it) after the class body is executed wouldn't be the worst option as long as we could be confident that code that doesn't pass lint won't raise an error before it gets to lint.

leonardt, 2020-09-22:
Also, I remembered that the dict_lookup helper we added (https://github.com/phanrahan/magma/blob/4c1a7cbcd6f9f4cb9e96b7c8d0adece0b7bba382/docs/primitives.md#dict-and-list-lookup) is very similar to the Case function you're describing so users do have that option (the main difference is dict_lookup acts more like a traditional case statement where a single condition is checked against possible values [the keys in the dictionary].  we could provide a similar interface for providing general conditions for the keys, which would provide the interface you have shown).  One "real" example of where this approach is quite a bit more verbose than the when statement approach can be seen here:
* chisel router: https://github.com/ucb-bar/chisel-tutorial/blob/release/src/main/scala/examples/Router.scala#L65-L83
  In this example, the enq/deq methods are used in different cases of nested when statements.  Under the hood, these methods just assign (wire) values to the ready/valid ports and use conditional assignment logic from the when statement to insert the muxes.
* magma router: https://github.com/leonardt/magma_examples/blob/master/magma_examples/router.py#L47-L79
  In the magma version, we have to explicitly construct the logic for all the ready/valid ports, I'm not sure the case function would help us out here.  To me, the chisel version is significantly more readable.

rsetaluri, 2023-06-16:
Implemented by `m.when`!

closed 2023-06-16 by rsetaluri