phanrahan/magma #860: Extending Product Types

leonardt, 2020-10-08:
Scala/Chisel provides a pattern for extending bundles, so you can define a bundle with common port(s) and then extend them with specific ports.  Here's a simple example where a two types share a common `resp` port.
```scala
class NastiResponseChannel(implicit p: Parameters) extends NastiSlaveToMasterChannel()(p) {
  val resp = UInt(width = nastiXRespBits)
}

class NastiWriteResponseChannel(implicit p: Parameters) extends NastiResponseChannel()(p) {
  val id   = UInt(width = nastiWIdBits)
  val user = UInt(width = nastiBUserBits)
}

class NastiReadDataChannel(implicit p: Parameters) extends NastiResponseChannel()(p)
    with HasNastiData {
  val id   = UInt(width = nastiRIdBits)
  val user = UInt(width = nastiRUserBits)
}
```

While this pattern is particularly useful in scala/chisel based on how they handle parameters (this is where the reuse comes form), I was wondering if this was something we would want to enable in magma (and hwtypes).  I tried doing this an encountered an error:

```python
# a.py
import magma as m

class A(m.Product):
    x = m.In(m.Bits[5])

class B(A):
    y = m.In(m.Bit)

print(B)
```

```shell
base ❯ python a.py
Traceback (most recent call last):
  File "a.py", line 8, in <module>
    class B(A):
  File "/Users/lenny/repos/magma/magma/tuple.py", line 566, in __new__
    return super().__new__(mcs, name, bases, namespace, cache, **kwargs)
  File "/Users/lenny/repos/hwtypes/hwtypes/adt_meta.py", line 149, in __new__
    return mcs._cache_handler(cache, fields, name, bases, ns, **kwargs)
  File "/Users/lenny/repos/hwtypes/hwtypes/adt_meta.py", line 175, in _cache_handler
    t = mcs._from_fields(fields, name, bases, ns, **kwargs)
  File "/Users/lenny/repos/magma/magma/tuple.py", line 583, in _from_fields
    t = TupleKind.__new__(mcs, name, bases, ns, fields=tuple(fields.values()),
  File "/Users/lenny/repos/magma/magma/tuple.py", line 66, in __new__
    raise TypeError(
TypeError: Can't inherit from multiple different bound_types
```

and for hwtypes

```python
# b.py
import hwtypes as ht

class C(ht.Product):
    x = ht.BitVector[5]

class D(C):
    y = ht.Bit

print(D)
```

```shell
base ❯ python b.py
Traceback (most recent call last):
  File "b.py", line 8, in <module>
    class D(C):
  File "/Users/lenny/repos/hwtypes/hwtypes/adt_meta.py", line 149, in __new__
    return mcs._cache_handler(cache, fields, name, bases, ns, **kwargs)
  File "/Users/lenny/repos/hwtypes/hwtypes/adt_meta.py", line 175, in _cache_handler
    t = mcs._from_fields(fields, name, bases, ns, **kwargs)
  File "/Users/lenny/repos/hwtypes/hwtypes/adt_meta.py", line 485, in _from_fields
    t = super().__new__(mcs, name, bases, ns, **kwargs)
  File "/Users/lenny/repos/hwtypes/hwtypes/adt_meta.py", line 149, in __new__
    return mcs._cache_handler(cache, fields, name, bases, ns, **kwargs)
  File "/Users/lenny/repos/hwtypes/hwtypes/adt_meta.py", line 181, in _cache_handler
    return super().__new__(mcs, name, bases, ns, **kwargs)
  File "/Users/lenny/repos/hwtypes/hwtypes/adt_meta.py", line 263, in __new__
    raise TypeError("Can't inherit from multiple different bound_types")
TypeError: Can't inherit from multiple different bound_types
```

I think the one value in this pattern is the common parent type for the two response channels (can introspect if it's an instance of response channel without caring which one specifically).  We can do this now by simply leaving the parent type unbound and binding the types in the children (but then we have to repeat port definition, and it doesn't quite capture the type hierarchy as desired, since ideally the shared port is captured in the parent type so we can safely perform operations on that port).

@cdonovick do you have a sense of how difficult it would be to support a pattern like this in the current implementation? I'm also open to alternative ways to do this that might be more python/magma idiomatic.

cdonovick, 2020-10-08:
This is a pattern I have wanted to support for a while.  I am not sure how hard it would be to get to work.

rsetaluri, 2020-10-12:
What's the fundamental complexity here? Looking at the `Tuple.__new__` code (in magma at least), I don't see why having 2 bound bases would actually cause an ambiguity.

Probably I'm just not understanding it. But doesn't `dataclass` inheritance pose the same problem? It does work in that case.

cdonovick, 2020-10-14:
My recollection of the issues with supporting this were:

A) preventing nonsensical stuff like:
```Python
class Foo(Product, Sum): ...
```

B) Resolving the "order" of fields:
```Python
class A(Product):
   x = T

class B(Product):
   y = S

class C(A, B):
    z = R

# < for issubclass
assert C < Tuple[T, S, R]
```

C) Dealing with conflicting field types:
```Python
class A(Product):
   x = T

class B(A):
   x = S
```

D) resolving implicit type relationships (the biggest issue):
```Python
# Currently we have:
Class A(Product):
   x = T
   y = S

assert A < Tuple[T, S]

class B(Product):
   x = T
   y = S
   z = R

assert B < Tuple[T, S, R]

assert not Tuple[T, S, R] < Tuple[T, S]
# although it could be if we drop __len__ for the interface (to maintain behavioral subtyping)
```

Now if naively just allow extension types by removing my block and assume everything just works (it probably doesn't) we get the following behavior:
```Python
class C(A):
   z = R

assert C < Tuple[T, S]
assert C < Tuple[T, S, R] 
```
Which I would find a bit surprising unless the following was true:
```Python
assert Tuple[T, S, R] < Tuple[T, S]
# and ideally
assert Sum[T, S] < Sum[T, S, R]
```
To support this I think we would need to change how subtyping is resolved.  Currently the type machinery injects the implicit bases into the `__bases__` of a type then lets normal python handle the rest.  Now we could write custom `__instancecheck__` / `__subclasscheck__` which handle the logic of determining if a class is an implicit subclass.  However, this would require a rewrite of `Product` / `TaggedUnion` as they leverage `Tuple` / `Sum` to actually handle much of the work.  

For example instead maintaining a mapping of field names to fields `Product` stores a mapping of fields names to field indices.  So `a = A(); a.x` is resolved as `a[0]` which is then handled by (with some abuse of notation) `Tuple[T, S](a)[0]`.  This convenient because we do allow `a[0]` as `A < Tuple[T, S]`.

rsetaluri, 2020-10-14:
Ok, thanks for the explanation. @cdonovick if my understanding is correct, the issue is not that the machinery to make extending the tuples itself is hard, but that supporting the desired type hierarchy relationships in the presence of extension requires bunch of changes and decisions on what those new relationships are?

And to take that one step further, the core conceptual question is whether or not `Tuple[T1, ..., Tn] < Tuple[T1, ..., Tn, Tn+1]`?  Abstractly what would be bad about this? Haven't thought about this but wondering what weird implications this might have...

cdonovick, 2020-10-15:
@rsetaluri There is also the issue that actual implementation of product (at least on the hwtype side) relies on it subclassing tuple.

> Tuple[T1, ..., Tn] < Tuple[T1, ..., Tn, Tn+1]

The other way around.  Everything you can do to `R=Tuple[T1, ..., Tn]` you can do to `S=Tuple[T1, ..., Tn, Tn+1]`.  But not the other way around  so `S` is a *more* specific type than `R`, and hence `S < R`.  (With the caveat about `__len__`).  

> Abstractly what would be bad about this?

I think it would actually be a good thing.  A register interface with async reset would be a subtype of a register interface without the reset.  This is probably good as any function that expects a register without async reset should probably work just fine on one with a reset (assuming the interfaces are otherwise the same).

rsetaluri, 2020-10-15:
Ok think all this makes sense to me. Thanks for clarification.