phanrahan/magma #871: StagedMemoryPort should support operators

leonardt, 2020-10-21:
```python
import magma as m

class Foo(m.Circuit):
    io = m.IO(idx=m.In(m.UInt[2]), O=m.Out(m.UInt[4]))
    memory = m.Memory(4, m.UInt[4])()
    read_data = memory[io.idx]
    io.O @= read_data >> 2
```

results in 

```python
Traceback (most recent call last):
  File "mem.py", line 4, in <module>
    class Foo(m.Circuit):
  File "mem.py", line 8, in Foo
    io.O @= read_data >> 2
TypeError: unsupported operand type(s) for >>: 'StagedMemoryPortUInt[4]' and 'int'
```

The StagedMemoryPort logic was added so we could use `__getitem__` for both reads and writes (i.e. `mem[write_addr] @= write_data` and `read_data = mem[read_addr]` (rather than use `__setitem__` for writes, which would make mem wiring inconsistent with normal wiring (mem uses `=` whereas everything else uses `@=`).  I think we need to pass through operators to the underlying magma value, like we do with the sequential register wrapper.

rsetaluri, 2020-10-21:
I think this is an interesting question of wrapping types using protocol vs inheriting from types. Would something like this work?

```python
def derive_from(T):

    class _Derived(T):
        def override_this_fn(self, ...):
            ...

    return _Derived
```

I have a feeling it's not this easy but I'm not sure why.

rsetaluri, 2020-10-21:
Another option would be to create a base Protocol class which handles all the logic of passing through operators. And then each of these protocols could derive from that. So it would be like

```python
PassThroughProtocolMeta(MagmaProtocolMeta):
    ...

PassThroughProtocol(metaclass=PassThroughProtocolMeta):
    ...

# Glossing over metaclasses for following classes...
SeqRegWrapper(PassThroughProtocol): pass
StagedMemoryPort(PassThroughProtocol): pass
```