phanrahan/magma #884: Verilog source code line number to Magma source code file name and line number?

splhack, 2020-11-04:
nextpnr reports critical path with Verilog source code line number. https://github.com/YosysHQ/nextpnr/pull/494
Is it possible to get Magma source code file name and line number from the Verilog source code line number?
For example, generating Verilog-->Magma source map file like JavaScript-->TypeScript source map.

splhack, 2020-11-05:
ok, figured out `MAGMA_CODEGEN_DEBUG_INFO` env and `config.set_debug_mode`.
Would be nice if `ast_tools` could keep the original filename and line number?

test.py

    import os

    os.environ["MAGMA_CODEGEN_DEBUG_INFO"] = "True"

    import magma as m

    m.config.set_debug_mode(True)

    class top(m.Circuit):
        io = m.IO(btn_1=m.In(m.Bit), led_0=m.Out(m.Bit), led_1=m.Out(m.Bit),)

        @m.inline_combinational()
        def logic():
            if io.btn_1:
                io.led_0 @= m.bit(0)
                io.led_1 @= m.bit(1)
            else:
                io.led_0 @= m.bit(1)
                io.led_1 @= m.bit(0)

    m.compile("test", top, output="coreir-verilog", inline=True)

Output Verilog.

    module Mux2xOutBit (
        input I0,
        input I1,
        input S,
        output O
    );
    // Module `Mux2xOutBit` defined at magma/magma/generator.py:77
    reg [0:0] coreir_commonlib_mux2x1_inst0_out;
    // Instance `coreir_commonlib_mux2x1_inst0` created at .ast_tools/logic1436591935413.py:6
    always @(*) begin
    if (S == 0) begin
        coreir_commonlib_mux2x1_inst0_out = I0;
    end else begin
        coreir_commonlib_mux2x1_inst0_out = I1;
    end
    end

    assign O = coreir_commonlib_mux2x1_inst0_out[0];
    endmodule

    module top (
        input btn_1,
        output led_0,
        output led_1
    );
    // Module `top` defined at test.py:10
    // Instance `Mux2xOutBit_inst0` created at .ast_tools/logic1436591935413.py:6
    // Connection `(Mux2xOutBit_inst0.I0, bit_const_1_None_out)` created at .ast_tools/logic1436591935413.py:6
    // Connection `(Mux2xOutBit_inst0.I1, bit_const_0_None_out)` created at .ast_tools/logic1436591935413.py:6
    // Connection `(Mux2xOutBit_inst0.S, btn_1)` created at .ast_tools/logic1436591935413.py:6
    Mux2xOutBit Mux2xOutBit_inst0 (
        .I0(1'b1),
        .I1(1'b0),
        .S(btn_1),
        .O(led_0)
    );
    // Instance `Mux2xOutBit_inst1` created at .ast_tools/logic1436591935413.py:8
    // Connection `(Mux2xOutBit_inst1.I0, bit_const_0_None_out)` created at .ast_tools/logic1436591935413.py:8
    // Connection `(Mux2xOutBit_inst1.I1, bit_const_1_None_out)` created at .ast_tools/logic1436591935413.py:8
    // Connection `(Mux2xOutBit_inst1.S, btn_1)` created at .ast_tools/logic1436591935413.py:8
    Mux2xOutBit Mux2xOutBit_inst1 (
        .I0(1'b0),
        .I1(1'b1),
        .S(btn_1),
        .O(led_1)
    );
    // Connection `(led_0, Mux2xOutBit_inst0_O)` created at .local/lib/python3.8/site-packages/ast_tools/passes/util.py:208
    // Connection `(led_1, Mux2xOutBit_inst1_O)` created at .local/lib/python3.8/site-packages/ast_tools/passes/util.py:208
    endmodule

leonardt, 2020-11-05:
Yes, we can collect this information in debug mode.  The only downside is the current implementation is rather slow (need to call `inspect` module every time something is instanced or wired.  We can also improve it to dump a mapping file like you want, is there a specific format we should should use for this mapping?

I will look into the mapping from the ast_tools generated code.  @cdonovick is working on integrating a new version of this code that uses libcst, which may make this easier.  Perhaps we should just make it modular and have a `python_source -> ast_tools_file` mapping and then "join" that with the `ast_tools_file -> verilog` mapping (that we currently could produce).

splhack, 2020-11-05:
Chisel's source locators format is good enough?
https://stackoverflow.com/questions/53417212/how-to-decipher-comments-in-generated-verilog-from-chisel

JavaScript source map may be too complicated.
https://sourcemaps.info/spec.html