phanrahan/magma #898: Magma should raise an exception when a design contains wiring errors

leonardt, 2021-01-20:
Here's an example of a circuit that has a wiring error, but will compile
```python
class T(m.Product):
    a = m.In(m.Clock)
    b = m.In(m.Clock)

class Foo(m.Circuit):
    io = m.IO(I=In(T), O=m.Out(m.Clock))
    m.wire(io.I, io.O)

m.compile("build/Foo", Foo)
```
This will produce the following error output:
```
ERROR:magma:Cannot wire Foo.I (Tuple(a=Out(Clock),b=Out(Clock))) to Foo.O (In(Clock))
```

However, it also generates a verilog file since the backend automatically wires up the undriven clock (Foo.O) to a default driver and compiles fine through coreir.

We know our users are not always the best at reading error messages, especially when their code contains a bunch of warnings that might hide the errors.

I propose that for each circuit definitions, we track the number of logged errors.  Then, when compiling a design we raise an exception at the Python level if the number of errors is non zero for all the circuits in a design.  We could then report all the errors that have been collected.

Another useful mode might be to have errors scoped per circuit.  So, at the end of each circuit definition, check if any errors have been logged and raise an error if so.  The idea here is that while we may not want to stop on every error (so we can collect multiple errors and report them at once, similar to how many software compilers will report more than one error if possible), but it may be useful to limit multiple errors on a per-circuit basis (since if there are multiple errors in a circuit, we probably want to fix those first before looking at other errors in the design).

leonardt, 2021-01-20:
Another niche case where this would have been helpful.

We had a bug in a circuit that was raising a wiring error:
```python
---------------------------------------------- Captured stderr call ----------------------------------------------
ERROR:magma:Cannot wire ..O (Out(SInt[16])) to .O (In(UInt[16]))
ERROR:magma:Output port ALU.O not driven
----------------------------------------------- Captured log call ------------------------------------------------
ERROR    magma:logging.py:91 Cannot wire ..O (Out(SInt[16])) to .O (In(UInt[16]))
ERROR    magma:logging.py:91 Output port ALU.O not driven
```

however, when the test is run in a suite, the error is reported the first time a circuit is used and defined.  The first time it was used and defined, the test actually passed, so this error output was surpressed.  In a subsequent test, this missing driver caused an error in the backend, but the error log wasn't reported because it was already outputted during a previous test.  Here we should've raised an exception during compile that found a circuit with errors and reported the errors (so they appear in the test output log).  The way we discovered it was running the test in isolation (rather the suite) and then seeing the log reported in that case.

rsetaluri, 2021-01-21:
1. I agree that it's weird that it's possible to have an error in the course of a design, but then it can still compile. However; I will argue that give the nature of the DSL, it's not that weird. Specifically, magma circuits are built up procedurally. As an example, imagine that we have a circuit definition for which we partially wire up the outputs. Then later, we wire up the remaining outputs using `open()`. The circuit may produce an `"unwired outputs"` error first, but it may later be a valid/complete circuit which can be compiled. I would admit that leaving outputs unwired is a bit different than having incorrect wiring statements (due to typing), but even in the latter case, since it's not a "fatal" error, the circuit is left unchanged, the same as not doing the wiring at all.

Ultimately, I think there are many combinations of (a) when errors are reported, (b) whether/what errors are fatal, and (c) how forgiving the compiler is to incomplete/incorrect circuit descriptions. The requested change is just one spot in this space. While I don't think it's nec. bad, I think it would be prudent to step back and settle on a set of rules across all of these concerns. (I can imagine that the above scenario is actually *not* desired by some design teams.)

2. Agree with this case pretty universally. Also, any idea why we're still getting `..O` kind of messages? It should've been fixed in a previous PR - can you mention the test so I can check it out?

commit d8ff4ef3 (2021-02-24): Merge pull request #917 from phanrahan/catch-error-log

[Fix #898] Raise exception when circuit has errors
commit e79073b6 (2021-02-24): [Fix #898] Raise exception when circuit has errors

This adds logic to the staged logger flow to store whether any error
messages were recorded during the circuit definition.

If an error was encountered, the backend will raise an exception rather
than proceeding.

This should avoid difficult to parse errors in the backend when
operating on a malformed circuit.  This should also hopefully surface
the error message to the user without less visual noise (e.g. a
traceback from an exception in the backend, leading them to believe it
could be an internal compiler error or something).

The hope is t [... cut]
closed 2021-02-24 by rsetaluri