phanrahan/magma #905: Failed to pass pytest after installation

griffin-warrior, 2021-02-01:
Hi,

Strictly followed the installation instruction, I have tried several python versions inside miniconda,  from 3.6 to 3.8, yet the installation still can't pass the pytest. FYI, I'm using Ubuntu 18.04.
For error messages, please check out the attached log file.
[pytest_error_messages.txt](https://github.com/phanrahan/magma/files/5904003/pytest_error_messages.txt)

leonardt, 2021-02-01:
In the log I see a g++ error coming from verilator repeated:
```
g++: error: unrecognized command line option '-faligned-new'; did you mean '-falign-jumps'?
```

It looks like there's an incompatibility with your versions of verilator and g++, what versions are you using?

leonardt, 2021-02-01:
Related: https://github.com/pytorch/pytorch/issues/9208

It looks like a gcc version issue.  You may want to try reporting this to the verilator community as it's a specific problem with the flags they're using with gcc, rather than issue with magma

griffin-warrior, 2021-02-02:
> In the log I see a g++ error coming from verilator repeated:
> 
> ```
> g++: error: unrecognized command line option '-faligned-new'; did you mean '-falign-jumps'?
> ```
> 
> It looks like there's an incompatibility with your versions of verilator and g++, what versions are you using?

gcc (Ubuntu 6.5.0-2ubuntu1~18.04) 6.5.0 20181026
Verilator 3.916 2017-11-25 rev verilator_3_914-65-g0478dbd

Thanks for your quick reply. I'm gonna try different GCC versions to see if this can be fixed.

griffin-warrior, 2021-02-02:
@leonardt Your tip works for me. I just switched GCC/G++ to the default version 'gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0', and all the errors were gone! But there still exists some skipped items. Do they matter?

===== 789 passed, 6 skipped in 139.70s (0:02:19) ======

leonardt, 2021-02-02:
Those should be fine to ignore.  Some tests are skipped by default if they can't find some optional dependency (e.g. a non-standard library that isn't required for core magma).  They are only necessary if you plan to use that feature but for now you should be fine to get started.

griffin-warrior, 2021-02-03:
@leonardt Got it. Thanks. :)

closed 2021-02-03 by griffin-warrior