phanrahan/magma #936: Scope not passed through to function calls in inline_combinational?

rsetaluri, 2021-03-08:
Having trouble with this code:

```python
import magma as m

class Main(m.Circuit):
    io = m.IO(O=m.Out(m.Bit), I=m.In(m.Bit))

    flag = io.I

    def get_flag():
        return io.I

    @m.inline_combinational(debug=True, file_name="inline_comb.py")
    def _():
        # This works:
        # if flag:
        # This doesn't work:
        if get_flag():
            io.O @= io.I
        else:
            io.O @= ~io.I
```

Traceback is:
```
Traceback (most recent call last):
  File "tmp.py", line 4, in <module>
    class Main(m.Circuit):
  File "tmp.py", line 12, in Main
    @m.inline_combinational(debug=True, file_name="inline_comb.py")
  File "/private/tmp/magma/env/lib/python3.7/site-packages/ast_tools/passes/util.py", line 208, in __call__
    return self.exec(etree, stree, env, metadata)
  File "/private/tmp/magma/magma/syntax/inline_combinational.py", line 117, in exec
    result = fn()
  File ".ast_tools/inline_comb.py", line 5, in _
    _cond_0 = get_flag()
  File "tmp.py", line 10, in get_flag
    return io.I
NameError: name 'io' is not defined
```

The comb file python generated is:
```python
def _():
    # This works:
    # if flag:
    # This doesn't work:
    _cond_0 = get_flag()
    _auto_prefix_0_0__0 = io.I
    _auto_prefix_0_0__1 = ~io.I
    _auto_prefix_0_0__2 = __phi(_cond_0, _auto_prefix_0_0__0, _auto_prefix_0_0__1)
    __0_return_0 = _auto_prefix_0_0__2
    return __0_return_0
```

Which looks right

leonardt, 2021-03-08:
Probably related to this: https://github.com/phanrahan/magma/issues/812

leonardt, 2021-03-08:
Actually this might be a python level problem, this code doesn't work:

```
class Foo:
    x = 3

    def get_flag():
        print(f"x = {x}")

    def foo():
        get_flag()

Foo.foo()
```
produces
```
Traceback (most recent call last):
  File "a.py", line 11, in <module>
    Foo.foo()
  File "a.py", line 8, in foo
    get_flag()
NameError: name 'get_flag' is not defined
```
The reason it works in computational might be because we're explicitly passing locals to the environment.

If I make it a classmethod, I then get an error similar to above

```
class Foo:
    x = 3

    def get_flag():
        print(f"x = {x}")

    @classmethod
    def foo(cls):
        cls.get_flag()

Foo.foo()
```

produces

```
Traceback (most recent call last):
  File "a.py", line 12, in <module>
    Foo.foo()
  File "a.py", line 9, in foo
    cls.get_flag()
  File "a.py", line 5, in get_flag
    print(f"x = {x}")
NameError: name 'x' is not defined
```

leonardt, 2021-03-08:
The issue seems to be that the method doesn't lookup names in the class namespace without the explicit class handle

leonardt, 2021-03-08:
Seems like the only way around this would be to define a combinational classmethod (which could then call get_flag as a classmethod).  This could be a decent option since inline_combinational has a couple quirks to it, so making it a classmethod with explicit references would avoid these scoping issues given that the users can explicitly refer to the class scope with `cls`, but it does add another step/level of indirection

cdonovick, 2021-03-08:
Might be able to hack around by explicitly passing  an env https://github.com/leonardt/ast_tools/blob/master/tests/test_passes.py#L129

cdonovick, 2021-03-08:
Oh no lenny is right this just is never going to work in python

cdonovick, 2021-03-08:
(without making it a cls method)

rsetaluri, 2021-03-08:
Good catch. I think making inline_combinational a class method isn't unreasonable but it's a pretty niche use-case so not itching to have it be supported.

In case you're interested, the reason this came up is because I was thinking of how to mix meta-programmed magma with `inline_combinational`. It seems like function calls are a good way to do this, since they're not touched by the SSA transform, but can use values appropriately. For exmaple

```python
def chain(l, size):
    for i in range(size - 1):
        l[i].I @= l[i + 1].O

class Top(m.Circuit):
    io = m.IO(I=m.In(m.Bit), flag=m.In(m.Bit))
    size = 64
    array = [m.Register(...)() for _ in range(size)]

    @m.inline_combinational
    def _():
        if io.flag:
            array[size - 1] @= <something>
            chain(array, 64)
        else:
            ...
```