phanrahan/magma #941: Assignment for wiring ports in magma sequential

cdonovick, 2021-03-19:
It is currently difficult to create wiring loops in magma sequential as the output of a circuit cannot be probed prior to calling it.

One solution to this would be to instead of using call to perform a wire allow assignment of ports. 

e.g.:
```Python3
def __call__(self):
    out = self.reg.out
    self.reg.in = out + 1
    return out
```
While this specific example can be solved using the `prev` construct in general it is difficult to construct such wiring loops.

leonardt, 2021-03-19:
Seems reasonable, I'm guessing we use "last assign" semantics.  In the case of branches, we probably need to require a default value in general (and raise an exception if there's a case where the port isn't assigned).  Are there any other little issues like this we should consider?

cdonovick, 2021-03-19:
I would handle the assignment in SSA to ensure only one value is ever / always assigned.   In strict mode it would raise an exception if a port was not assigned in a path, without strict it would assume last assigned value is the default (similar to the handling of return values)

leonardt, 2021-06-07:
Just for reference, I think the current plan is to use a wiring style syntax rather than assignment, see https://github.com/phanrahan/magma/pull/958