phanrahan/magma #942: [Feature Request] user-named outputs for circuit.sequential2

rdaly525, 2021-03-23:
Currently output names are generated as "O" for a single output and  "O0" .. "O<N-1>" for multiple outputs. It would be great if we could add an ability to name the outputs ports similar to naming the input ports.

Possible APIs (in order of my personal preference)

1)
```
    @m.sequential2()
    class A:
        ...
        def __call__(self, i0: m.Bit, i1: m.Bit) -> {"myport0" : m.Bit, "myport1": m.Bit}:
            ...
```

2) (Current one that Peak supports)
```
    @m.sequential2()
    class A:
        ...
        @name_outputs(myport0=m.Bit, myport1=m.Bit)
        def __call__(self, i0: m.Bit, i1: m.Bit):
            ...
```

3)
```
    class Outputs(m.Product):
        myport0=m.Bit
        myport1=m.Bit

    @m.sequential2()
    class A:
        ...
        def __call__(self, i0: m.Bit, i1: m.Bit) -> Outputs
            ...
```
This one could cause some ambiguity if you actually want a single port which is itself a Product type.

I am happy to look into implementing. @leonardt, thoughts?

rsetaluri, 2021-03-23:
@rdaly525 why can't we use the `@name_outputs` decorator already around?

rdaly525, 2021-03-23:
@rsetaluri, are you saying that name_outputs decorator already currently works in magma or are you advocating that API 2 would be your preference

rsetaluri, 2021-03-23:
Oops I think I misinterpreted. So `@name_outputs` works, its just that magma ignores it right now?

In that case, yes, I would advocate for (2). (Though I'm not a huge stakeholder here, I just remember working with (2) 😄.)

rdaly525, 2021-03-23:
so option 4?:

```
    @m.sequential2()
    class A:
        ...
        @name_outputs("myport0", "myport1")
        def __call__(self, i0: m.Bit, i1: m.Bit) -> (m.Bit, m.Bit):
            ...

```

rdaly525, 2021-03-23:
I suspect that it would be easiest to implement 1, and have both 2 and 4 be syntax sugar for 1

rsetaluri, 2021-03-23:
Ignoring the (2) vs (4) thing, it seems easiest to do (2) no? Since the feature is already implemented? It just requires magma to inspect that field and create the right output names?

I also like that it doesn't muddy the function signature and is orthogonal to the signature, i.e. it's trivial to allow it to not be specified, and also trivial to inspect whether or not it's been specified.

cdonovick, 2021-04-06:
@rsetaluri @leonardt is this a feature you want "natively" in magma?

Implementation wise I think it would be easiest to support in the following way:
```Python
@m.sequential2(output_names=['myport0', 'myport1'])
class A:
    ...
    def __call__(self, i0: m.Bit, i1: m.Bit) -> (m.Bit, m.Bit):
        ...
```

And just have peak  pass the correct `output_names` based on `name_outputs`

leonardt, 2021-04-06:
That seems reasonable to me.

Another option is something in the declaration like
```
def __call__(...) -> m.Output(myport0=m.Bit, myport1=m.Bit)
```

I was thinking like m.IO where we can use kwargs for name, but we're not handling inputs, but m.O seems odd.  m.Out would be better but it overloads the type qualifier which I think we may want to avoid.  m.Ports? the decorator seems fine for now though.

rsetaluri, 2021-04-06:
I'd prefer the decorator. No preference between (a) `output_names` keyword-arg on `m.sequential2`, or (b) `@name_outputs` decorator on `__call__` directly. But I wouldn't want both to exist. If we do (a), then we should deprecate (b) (since it's live right now).

cdonovick, 2021-04-06:
b) is a peak feature not something that works in magma (hence the issue).

splhack, 2021-04-06:
It already supported pretty much 1 with `AnonProduct`

https://github.com/phanrahan/magma/blob/3291f43933154a6f69789320fb491fbcd6de7332/tests/test_syntax/test_sequential2.py#L288

But I feel `AnonProduct(dict(` is still long to write.

https://github.com/phanrahan/magma/pull/568, miss `m.Tuple`

    def test(I: m.Bits[2]) -> m.Tuple(x=m.Bit, y=m.Bit):

cdonovick, 2021-04-07:
This slightly different as it requires an explicit cast to a product type.  The idea here is you can have normal returns with custom output port names
```Python
@m.sequential2(output_names=['myport0', 'myport1'])
class A:
    def __call__(self, i0: m.Bit, i1: m.Bit) -> (m.Bit, m.Bit):
        return i0, i1

assert A.interface.ports.keys() == {'i0', 'i1', 'myport0', 'myport1', 'CLK'} 
```

PR #947 (2021-04-07): Allow seq outputs to be named
closes #942
closed 2021-04-07 by leonardt