phanrahan/magma #965: Add ability to log automatically wired clocks

leonardt, 2021-06-08:
Should be off by default, and should be able to dump to file rather than stdout since that would likely be too verbose.  Could be useful to debugging clock related wiring errors (can scan the log to see where clocks are being automatically wired).  Useful for designs that use manual clock wiring (can enable log or error to detect clocks that aren't wired manually).

rsetaluri, 2021-06-09:
IMO, we should put it in the normal logging flow and have the ability to log to file generally. it is pretty clunky/hard to maintain to have super fine grain control over what gets logged to file vs what gets logged to stdout. we should do something more general. can take this on.

leonardt, 2022-03-31:
Done

closed 2022-03-31 by leonardt