phanrahan/magma #967: Forgetting parens (instance) when using Generator2 provides a confusing error

leonardt, 2021-06-08:
Here's an example:
```python
def test_missing_parent():
    class Foo(m.Generator2):
        def __init__(self, width):
            self.io = m.IO(x=m.In(m.Bits[width]), y=m.Out(m.Bits[width]))
            self.io.y @= self.io.x

    class Bar(m.Generator2):
        def __init__(self, width):
            self.io = m.IO(x=m.In(m.Bits[width]), y=m.Out(m.Bits[width]))
            # Missing a parent
            self.io.y @= Bar(width)(self.io.x)

    m.compile("build/Bar", Bar)
```

produces

```
========================================================== FAILURES ==========================================================
____________________________________________________ test_missing_parent _____________________________________________________

    def test_missing_parent():
        class Foo(m.Generator2):
            def __init__(self, width):
                self.io = m.IO(x=m.In(m.Bits[width]), y=m.Out(m.Bits[width]))
                self.io.y @= self.io.x

        class Bar(m.Generator2):
            def __init__(self, width):
                self.io = m.IO(x=m.In(m.Bits[width]), y=m.Out(m.Bits[width]))
                # Missing a parent
                self.io.y @= Bar(width)(self.io.x)

>       m.compile("build/Bar", Bar)

tests/test_generator2.py:110:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
magma/compile.py:55: in compile
    compiler = _make_compiler(output, main, basename, opts)
magma/compile.py:35: in _make_compiler
    return CoreIRCompiler(main, basename, opts)
magma/backend/coreir/coreir_compiler.py:78: in __init__
    self.deps = self._deps()
magma/backend/coreir/coreir_compiler.py:172: in _deps
    pass_ = InstanceGraphPass(self.main)
magma/passes/passes.py:104: in __init__
    pass_ = BuildInstanceGraphPass(main).run()
magma/passes/passes.py:63: in run
    self._run(self.main)
magma/passes/passes.py:75: in _run
    super()._run(circuit)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <magma.passes.passes.BuildInstanceGraphPass object at 0x7fa143f56190>
circuit = <class 'test_generator2.test_missing_parent.<locals>.Bar'>

    def _run(self, circuit):
        if isdefinition(circuit):
>           for inst in circuit.instances:
E           TypeError: 'property' object is not iterable

magma/passes/passes.py:53: TypeError
================================================== short test summary info ===================================================
FAILED tests/test_generator2.py::test_missing_parent - TypeError: 'property' object is not iterable
================================================ 1 failed, 6 passed in 0.13s =================================================
```

We should find a way to detect and traceback this error to the sourceline to make it obvious to the user what's wrong.

leonardt, 2021-06-10:
need to try to find a coreir segfault error

leonardt, 2021-06-10:
priority 2

leonardt, 2021-06-10:
Updated with a case that segfaults, basically forgetting a paren will wire up a value from a different circuit.  When compiling, this can create a coreir select that is invalid (e.g. trying to select an interface port that doesn't exist).  Will look into where in the magma code we can validate this.  I think we need to ensure that defn/inst refs are from the correct circuit context.  We could also try to catch this during wiring.

rsetaluri, 2021-06-11:
I think the most robust thing to do is to enforce certain things at wire time (though it is somewhat expensive, maybe can be in some debug mode only).

First let's define a few auxiliary functions:
* `inst :: m.Type -> Optional[m.Circuit]`: if the value is part of a port of an instance, return that instance. otherwise, return `None`.
* `defn :: m.Type -> Optional[m.CircuitKind]`: if the value is part of a part of a definition, return that definition. otherwise, return `None`.
* `temp :: m.Type -> bool`: returns whether this value is a temporary, basically `temp(x) === (inst(x) is None && defn(x) == None)`.
* `top :: void -> m.CircuitKind`: returns the top of the definition context stack (for now we ignore that it we don't actually have the CircuitKind object directly, but we have the placer object which is 1-to-1 so conceptually it's the same). also we assume, there is always a top (i.e. we are inside of a defn).
* `container :: m.Circuit -> m.CircuitKind`: returns the containing definition of this instance (same comment re: placer holds here).

Let's say you are trying to wire `a` to `b`. Furthermore, let's assume `a` and `b` are instance of `m.Bit`, without loss of generality, since we can just enforce this at the lowest bit-wiring level. We can always optimize it to work on partial or whole values; I just want to write down the most precise form of the condition.

Then, we want to enforce the following condition:

```python
  temp(a)
| temp(b)  # if either value is temporary we accept it. see note later.
| (defn(a) & defn(b) & (defn(a) == defn(b))  # defn-port to defn-port wiring
| (inst(a) & defn(b) & (container(inst(a)) == defn(b)) & (defn(b) == top())  # inst-port to containing-defn-port wiring, w/ additional constraint of top-ness.
| '                                                                         ' # same thing
```

leonardt, 2021-08-06:
Re: handling temporaries, there's actually some pretty complicated code in the repr flow (used mainly by uniquification) that reconstructs the "wires" (non anonymous temporaries) by tracing the values in the definition (see https://github.com/phanrahan/magma/blob/28427f93a28fbc480a8655f81a4f5f8151514375/magma/interface.py#L95).  I think we could improve the handle of temporaries in general to associate themselves with a definition context (just like an instance, they could peek on the definition context stack to find the association).  This would clean up that repr logic and address the issues with temporaries here.

For avoiding the tracing, we could do two things: (1) cache the trace, so we only pay for the traversal up once, (2) construct the children with a reference to their defn/inst (so pass this information down the tree as it's constructed, to avoid a lookup later).

leonardt, 2021-09-23:
Fixed by #990

closed 2021-09-23 by leonardt