phanrahan/magma #1020: Recent changes broke garnet

Kuree, 2022-03-04:
https://github.com/phanrahan/magma/commit/cb076eb5d4de94072d506d993b8ee6c288a6756f is causing errors in https://github.com/StanfordAHA/garnet/runs/5417103429?check_suite_focus=true#step:6:227

leonardt, 2022-03-04:
@rsetaluri can you take a look at this? I hotfixed this on master by just moving the old bitutils file to mantle and patching the functions in since they're used in a lot of places.  Not sure if we want to update all that old code since we removed a bunch of the functions being used, or just keep the old bitutils around there for backwards compat.

rsetaluri, 2022-03-04:
@leonardt thanks for patching. IMO we can keep this hack (I'll clean it up and actually move the functions into mantle). When we update to mantle2 I imagine these generators will go away anyway.

leonardt, 2022-03-04:
Closing this then.

closed 2022-03-04 by leonardt