phanrahan/magma #1073: Undriven port bug

Kuree, 2022-05-15:
Mamga reports errors of undriven ports even though it's connected.

How to reproduce:
Uinsg `magma-lang==2.2.18`
Remove this hack here:
https://github.com/StanfordAHA/canal/blob/37799cb4db4e5260ea538b8650c5446759e7892c/canal/circuit.py#L1090
To see the error.

To make the error more obvious, I wired to the same port multiple times. Here is the output when the hack is in place:
```
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_SOUTH_SB_OUT_B16.valid_in[0], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_NORTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_EAST_SB_OUT_B16.valid_in[0], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_NORTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_WEST_SB_OUT_B16.valid_in[0], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_NORTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_NORTH_SB_OUT_B16.valid_in[0], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_SOUTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_EAST_SB_OUT_B16.valid_in[1], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_SOUTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_WEST_SB_OUT_B16.valid_in[1], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_SOUTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_NORTH_SB_OUT_B16.valid_in[1], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_EAST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_SOUTH_SB_OUT_B16.valid_in[1], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_EAST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_WEST_SB_OUT_B16.valid_in[2], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_EAST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_NORTH_SB_OUT_B16.valid_in[2], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_WEST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_SOUTH_SB_OUT_B16.valid_in[2], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_WEST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_EAST_SB_OUT_B16.valid_in[2], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T0_WEST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_SOUTH_SB_OUT_B16.valid_in[0], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_NORTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_EAST_SB_OUT_B16.valid_in[0], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_NORTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_WEST_SB_OUT_B16.valid_in[0], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_NORTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_NORTH_SB_OUT_B16.valid_in[0], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_SOUTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_EAST_SB_OUT_B16.valid_in[1], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_SOUTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_WEST_SB_OUT_B16.valid_in[1], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_SOUTH_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_NORTH_SB_OUT_B16.valid_in[1], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_EAST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_SOUTH_SB_OUT_B16.valid_in[1], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_EAST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_WEST_SB_OUT_B16.valid_in[2], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_EAST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_NORTH_SB_OUT_B16.valid_in[2], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_WEST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_SOUTH_SB_OUT_B16.valid_in[2], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_WEST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_EAST_SB_OUT_B16.valid_in[2], Old Output: SB_ID0_2TRACKS_B16_ReadyValidCore.WIRE_SB_T1_WEST_SB_IN_B16.valid_out, New Output: VCC
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_NORTH_SB_OUT_B16.valid_in[0], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_NORTH_SB_OUT_B16.valid_in[1], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_NORTH_SB_OUT_B16.valid_in[2], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_SOUTH_SB_OUT_B16.valid_in[0], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_SOUTH_SB_OUT_B16.valid_in[1], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_SOUTH_SB_OUT_B16.valid_in[2], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_EAST_SB_OUT_B16.valid_in[0], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_EAST_SB_OUT_B16.valid_in[1], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_EAST_SB_OUT_B16.valid_in[2], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_WEST_SB_OUT_B16.valid_in[0], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_WEST_SB_OUT_B16.valid_in[1], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T0_WEST_SB_OUT_B16.valid_in[2], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_NORTH_SB_OUT_B16.valid_in[0], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_NORTH_SB_OUT_B16.valid_in[1], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_NORTH_SB_OUT_B16.valid_in[2], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_SOUTH_SB_OUT_B16.valid_in[0], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_SOUTH_SB_OUT_B16.valid_in[1], Old Output: VCC, New Output: GND
WARNING:magma:Wiring multiple outputs to same wire, using last connection. Input: MUX_SB_T1_SOUTH_SB_OUT_B16.valid_in[2], Old Output: VCC, New Output: GND
[... cut ...]
ERROR:magma:SB_ID0_2TRACKS_B16_ReadyValidCore.MUX_SB_T1_NORTH_SB_OUT_B16.valid_in not driven
ERROR:magma:SB_ID0_2TRACKS_B16_ReadyValidCore.MUX_SB_T1_SOUTH_SB_OUT_B16.valid_in not driven
ERROR:magma:SB_ID0_2TRACKS_B16_ReadyValidCore.MUX_SB_T1_EAST_SB_OUT_B16.valid_in not driven
ERROR:magma:SB_ID0_2TRACKS_B16_ReadyValidCore.MUX_SB_T1_WEST_SB_OUT_B16.valid_in not driven
```

You can see that the duplicated connection is properly detected yet the undriven logic can't pick up the correct connection. You should see similar output if the second redundant connection is removed.

Kuree, 2022-05-16:
It turns out that the undriven port is only *partially* connected. Shouldn't the error message give connection information on a bit-level rather than on a bus-level?

leonardt, 2022-05-17:
Yes, it should try to print out the children if it detects partial connection, I think there might be a regression from the array2 change, taking a look

leonardt, 2022-05-19:
Turn out it was logging this information but only at debug level.  I changed it to error since I think this should be reported by default when an error occurs: https://github.com/phanrahan/magma/pull/1076

commit 260d715d (2022-05-20): Fix #1073 (#1076)
closed 2022-05-20 by leonardt