phanrahan/magma #1185: m.compile(..., output="verilog") leads to AttributeError: 'AnonRef' object has no attribute 'tuple'

rkshthrmsh, 2022-11-16:
https://github.com/phanrahan/magma/blob/1d7b3b94b6dd7c33cc60b5b0f2d73dc2e2fb4894/magma/backend/verilog.py#L138

leonardt, 2022-11-16:
Hi @rkshthrmsh , the `verilog` output is no longer supported.  The preferred method of compiling is using "mlir-verilog", but it requires setting up and installing CIRCT, see these instructions https://github.com/llvm/circt#setting-this-up

Can you try using the mlir-verilog output instead?

rkshthrmsh, 2022-11-16:
Hi @leonardt, upon installing CIRCT (and the llvm/mlir) dependencies and attempting to compile to "mlir-verilog", I am seeing the following error:
`~/ws/magma_examples/magma_examples$ python3 byte_selector.py 

Traceback (most recent call last):

File "/home/giambla2/ws/magma_examples/magma_examples/byte_selector.py", line 21, in <module>

    m.compile("ByteSelector", ByteSelector, output="mlir-verilog")

  File "/home/giambla2/.local/lib/python3.9/site-packages/magma/compile.py", line 106, in compile

    result = compiler.compile()

  File "/home/giambla2/.local/lib/python3.9/site-packages/magma/backend/mlir/mlir_compiler.py", line 64, in compile

    mlir_to_verilog(fi, fo, opts=self._mlir_to_verilog_opts)

  File "/home/giambla2/.local/lib/python3.9/site-packages/magma/backend/mlir/mlir_to_verilog.py", line 117, in mlir_to_verilog

    returncode = _run_subprocess(cmd, istream, ostream)

  File "/home/giambla2/.local/lib/python3.9/site-packages/magma/backend/mlir/mlir_to_verilog.py", line 74, in _run_subprocess

    proc = subprocess.Popen(args, stdin=stdin_actual, stdout=stdout_actual)

  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__

    self._execute_child(args, executable, preexec_fn, close_fds,

  File "/usr/lib/python3.9/subprocess.py", line 1821, in _execute_child

    raise child_exception_type(errno_num, err_msg, err_filename)

FileNotFoundError: [Errno 2] No such file or directory: 'circt-opt'`

Circt-opt is available in the CIRCT build directory. Is a mamga configuration required to point to it?

leonardt, 2022-11-16:
I would suggest adding the circt build directory to your path, e.g.

```
export PATH=$PATH:/path/to/circt/build
```

rkshthrmsh, 2022-11-16:
Would building llvm/mlir and circt with python bindings enabled (as mentioned here: https://github.com/llvm/circt/blob/main/docs/PythonBindings.md#manual-compilation) be another option?

Updating the path worked, thanks!

closed 2022-11-17 by rkshthrmsh