phanrahan/magma #1190: Support for custom interfaces

rkshthrmsh, 2022-11-17:
What is the support (current or planned) for custom interfaces? Is it currently possible to declare an interface, embed it into a port list, and wire them? An example:
```python
import magma as m

axisSlave = m.Interface(
        'tdata', m.In(m.UInt[32]),
        'tvalid', m.In(m.Bit),
        'tready', m.Out(m.Bit)
        )
# Is this the correct way of defining an interface?

axisMaster = m.Interface(
        'tdata', m.Out(m.UInt[32]),
        'tvalid', m.Out(m.Bit),
        'tready', m.In(m.Bit)
        )

class AXISTest(m.Circuit):
    # Do circuit ports support custom interfaces?
    io = m.IO(
            AXIS_S=axisSlave(),
            AXIS_M=axisMaster()
    )

    # Is there a concise way of wiring the two interfaces?
    # Since the 'interface ports' are compatible
    # Something like
    # io.AXIS_M @= io.AXIS_S

    # or is this the way?
    io.AXIS_M.tdata @= io.AXIS_S.tdata
    io.AXIS_M.tvalid @= io.AXIS_S.tvalid

    io.AXIS_S.tready @= io.AXIS_M.tready

```
Finally, is higher order functional support available for them? Something like:

```python
class AXISWrapper(m.Circuit):
    io = m.IO(
            wrap_axis_s=axisSlave(),
            wrap_axis_m=axisMaster()
    )

    subs = [AXISTest() for _ in range(4)]
    io.wrap_axis_m @= m.fold(subs, foldargs={"AXIS_S": "AXIS_M"})(io.wrap_axis_s)
```

leonardt, 2022-11-17:
`m.Interface` is a part of the circuit data model, not a magma type.  Instead of `m.Interface`, I believe you are looking for `m.Product` which allows you to define a composite type with named fields.  Here's an example: https://github.com/leonardt/magma_examples/blob/ccfcbd800df3a613c8346ca7816df4ca96dbe134/magma_examples/router.py#L15

leonardt, 2022-11-17:
Products can be in bulk (i.e. you don't need to wire the fields)

leonardt, 2022-11-17:
And yes, you can use higher order operations on interfaces containing products.

rkshthrmsh, 2022-11-21:
Thank you, m.Product is what I was looking for.

closed 2022-11-21 by rkshthrmsh