phanrahan/magma #1192: Interpretation of @m.inline_combinational() usage

rkshthrmsh, 2022-11-18:
This circuit derived from the documentation (https://magma.readthedocs.io/en/latest/syntaxes/#inline-combinational) compiles into the verilog shown below. It appears that inline_combinational has generated a procedural block. Although the `automatic logic` variable will eventually synthesize into combinational logic, assignment to Register_inst0 is sequential. What is the correct interpretation of the 'combinational' in `@m.inline_combinational()`? Is it meant to signify that the function output within `logic()` is determined by the present inputs only, which in this case are `io.invert` and `reg.O`?

```python
class Main(m.Circuit):
    io = m.IO(invert=m.In(m.Bit), O0=m.Out(m.Bit))
    io += m.ClockIO()
    reg = m.Register(m.Bit)()

    @m.inline_combinational()
    def logic():
        if io.invert:
            reg.I @= ~reg.O
        else:
            reg.I @= reg.O

        io.O0 @= reg.O
```
```verilog
module Main(
  input  invert,
         CLK,
  output O0);

  reg Register_inst0;
  always_ff @(posedge CLK) begin
    automatic logic [1:0] _GEN;
    _GEN = {{~Register_inst0}, {Register_inst0}};
    Register_inst0 <= _GEN[invert];
  end // always_ff @(posedge)
  initial
    Register_inst0 = 1'h0;
  assign O0 = Register_inst0;
endmodule
```

leonardt, 2022-11-19:
Yes, inline combinational is meant to be similar to always_comb, however the way we generate the code obfuscates this a bit (we transform it to structural muxes via an ssa pass, the CIRCT/MLIR is inlining the mux logic into the always block).

I'd recommend using `m.when` over m.inline_combinational as this will generate `always_comb` logic and has support for more features.

rkshthrmsh, 2022-11-19:
Is there an `else` equivalent along with the `m.when` and `m.elsewhen`?

rkshthrmsh, 2022-11-19:
> I'd recommend using m.when over m.inline_combinational as this will generate always_comb logic and has support for more features.
Is your recommendation for all logic or specific to situations where the code contains sequential + combinational logic? I'm trying to understand if `m.inline_combinational` may be replaced / deprecated in favour of `m.when` in the near future.

leonardt, 2022-11-19:
m.otherwise is the equivalent to else, and yes we are planning to use m.when for the future and evaluate it, so far people are happy with it.

rkshthrmsh, 2022-11-19:
Great! Thanks, @leonardt.

closed 2022-11-19 by rkshthrmsh