phanrahan/magma #1193: How to improve the readability of generated verilog? Debug mode made no difference.

rkshthrmsh, 2022-11-18:
I am trying to make the compiled verilog more readable using the suggestions mentioned here: https://magma.readthedocs.io/en/latest/FAQ/#debugging-generated-verilog
```python
m.config.set_debug_mode(True)
m.set_codegen_debug_info(True)
```
Adding these configurations made no difference to the verilog between set_debug_mode set to True or False. How can I compile verilog in debug mode?

leonardt, 2022-11-19:
Unfortunately when we migrated to our MLIR backend we didn't include the debug code generation, we are working on that, see this relevant issue: https://github.com/phanrahan/magma/issues/1182

We should have this feature available soon.

rkshthrmsh, 2022-11-19:
I'll keep an eye out for it. Thanks!

closed 2022-11-19 by rkshthrmsh