phanrahan/magma #1196: What is the recommendation for FSMs in magma?

rkshthrmsh, 2022-11-21:
What is the recommended way for coding FSMs in magma? 

- `m.when` and `m.elsewhen` conditional statements
- `m.coroutine` / `m.sequential2`

The when and elsewhen constructs seem familiar from a verilog perspective albeit more analogous to if and else if. On the other hand, with coroutines it seems refactoring large FSMs with branches and loops would be easier and the code more readable.

- And is there support for case statements similar to `switch, is` in Chisel or an equivalent based on Python's `match, case`?

leonardt, 2022-11-21:
I would use `m.when` for now.  

We are planning to add `m.coroutine` style constructs using an `m.when` style syntax, but that feature is still planned and not ready.  We are working on a `sum` type feature tracked by this PR: https://github.com/phanrahan/magma/pull/1150  this introduces a match/case style syntax, which we then plan to add the coroutine style syntax on top of.

rkshthrmsh, 2022-11-21:
That sounds great. Thanks, @leonardt!

closed 2022-11-21 by rkshthrmsh