phanrahan/magma #1198: How to index with magma types and directions / pins

rkshthrmsh, 2022-11-22:
I am trying to design a circuit that takes a 16-bit data as an input, load it to a register and provide outputs that are based on an 'index' input. Here is the verilog equivalent:
```verilog
module index(
    input CLK,
    input [15:0] I,
    input load,
    input [3:0] index,
    output O1,
    output O2);
    reg [15:0] I_reg;
    wire [1:0] index1;
    always_ff @(posedge CLK) begin
        if (load)
            I_reg <= I;
    end
    assign index1 = index[1:0];
    assign O1 = I_reg[index];
    assign O2 = I_reg[index1];
endmodule
```
Here is the magma equivalent from my attempt:
```python
class Index(m.Circuit):
    io = m.IO(
            CLK=m.In(m.Clock),
            I=m.In(m.UInt[16]), 
            load=m.In(m.Bit),
            index=m.In(m.UInt[4]),
            O1=m.Out(m.Bit[1]),
            O2=m.Out(m.Bit[1])
    )
    I_reg = m.Register(m.UInt[16], has_enable=True)()(I=io.I, CE=io.load)
    index1 = io.index[:2]
    io.O1 @= I_reg[index1]
    io.O2 @= I_reg[io.index]
```
Which gives:
```python
Traceback (most recent call last):
  File "/home/giambla2/ws/magma_examples/magma_examples/test.py", line 223, in <module>
    class Index(m.Circuit):
  File "/home/giambla2/ws/magma_examples/magma_examples/test.py", line 230, in Index
    O1=m.Out(m.Bit[1]),
  File "/home/giambla2/.local/lib/python3.9/site-packages/magma/digital.py", line 59, in __getitem__
    raise TypeError('Direction of Digital must be an instance of '
TypeError: Direction of Digital must be an instance of m.Direction
```

How should this be implemented?

leonardt, 2022-11-22:
I think the issue here is that you're using `m.Bit[1]`, I think you either want to use `m.Bit` or `m.Bits[1]`

rkshthrmsh, 2022-11-22:
Thanks for catching that!

closed 2022-11-22 by rkshthrmsh