phanrahan/magma #1211: Does len() on magma enumerated type return bit length of member?

rkshthrmsh, 2022-12-20:
It appears that len() on magma enumerated type returns bit length of members. This differs from the native Python Enum, which returns the number of members. Bit length of member is useful, however, is there also a way to access the number of members?

```python
# Magma Enum
class State(m.Enum):
        IDLE=0
        CONFIG=1
        OPERATE=8

# len(State)
# 4
```
```python
# Python Enum
class State(Enum):
        IDLE=0
        CONFIG=1
        OPERATE=8

# len(State)
# 3
```

leonardt, 2022-12-20:
Added support for this via https://github.com/phanrahan/magma/pull/1212

rkshthrmsh, 2022-12-20:
Thanks! With this change, would State.size or State.N be the recommended way of accessing the bit length of the members?

leonardt, 2022-12-20:
It will expose access to an attribute "fields" which will contain a dictionary mapping to name to value, here's a simple example for what you're looking for:

```
    class MyEnum(m.Enum):
        x = 0
        y = 1

    assert len(MyEnum.fields) == 2
```

rkshthrmsh, 2022-12-20:
Alright. So, `len(MyEnum)` will return the bit length of any member, and `len(MyEnum.fields)` will return the number of members.

closed 2022-12-20 by rkshthrmsh