phanrahan/magma #1213: Using functions to assign values within an m.when style FSM

rkshthrmsh, 2022-12-20:
I am trying to create a somewhat simple FSM which uses a count  to update a register 'out'. The output of the register is connected to the circuit output 'O'. The register update value is a function of the count as defined in the 'config' function. It appears that the generated verilog code only evaluates the config function up to the first return statement, but nothing after that. So, the output O is always wired to the value 1 instead of becoming 5, 9, 13, 17. 

Is there a way to define functions in this manner? What would you recommend to use instead? Maybe related to #1196 

```python3
class State(m.Enum):
    IDLE = 0
    CONFIG = 1

def config(count):
    with m.when(count == 0):
        return m.uint(1, 32)
    with m.elsewhen(count == 1):
        return m.uint(5, 32)
    with m.elsewhen(count == 2):
        return m.uint(9, 32)
    with m.elsewhen(count == 3):
        return m.uint(13, 32)
    with m.elsewhen(count == 4):
        return m.uint(17, 32)

class Test(m.Circuit):
    io = m.IO(
        CLK=m.In(m.Clock),
        RESETN=m.In(m.ResetN),
        load=m.In(m.Bit),
        O=m.Out(m.UInt[32])
    )
    state = m.Register(m.UInt[len(State)], init=State.IDLE,
                       reset_type=m.ResetN)()
    count = m.Register(m.UInt[clog2(5)], reset_type=m.ResetN)()
    out = m.Register(m.UInt[32], reset_type=m.ResetN)()

    with m.when(state.O == State.IDLE):
        with m.when(io.load):
            state.I @= State.CONFIG
            count.I @= 0
    with m.elsewhen(state.O == State.CONFIG):
        with m.when(count.O < 5):
            out.I @= config(count.O)               # Issue here
            count.I @= count.O + 1
        with m.otherwise():
            state.I @= State.IDLE
            count.I @= 0
            out.I @= 0

    io.O @= out.O
```
```systemverilog
// Generated by CIRCT unknown git version
module Test(
  input         CLK,
                RESETN,
                load,
  output [31:0] O);

  reg [31:0] Register_inst2;
  reg        Register_inst0;
  reg [2:0]  Register_inst1;
  reg        _GEN;
  reg [2:0]  _GEN_0;
  reg [31:0] _GEN_1;
  always_comb begin
    _GEN = Register_inst0;
    _GEN_0 = Register_inst1;
    _GEN_1 = Register_inst2;
    if (~Register_inst0) begin
      if (load) begin
        _GEN = 1'h1;
        _GEN_0 = 3'h0;
      end
    end
    else if (Register_inst0) begin
      if (Register_inst1 < 3'h5) begin
        _GEN_1 = 32'h1;                           // Issue here. Not changing.
        _GEN_0 = Register_inst1 + 3'h1;
      end
      else
        _GEN = 1'h0;
    end
  end // always_comb
  always_ff @(posedge CLK) begin
    if (RESETN) begin
      Register_inst0 <= 1'h0;
      Register_inst1 <= 3'h0;
      Register_inst2 <= 32'h0;
    end
    else begin
      Register_inst0 <= _GEN;
      Register_inst1 <= _GEN_0;
      Register_inst2 <= _GEN_1;
    end
  end // always_ff @(posedge)
 
  assign O = Register_inst2;
endmodule
```

leonardt, 2022-12-20:
I would use wiring rather than returning, so something like this:
```
class State(m.Enum):
    IDLE = 0
    CONFIG = 1

def config(reg, count):
    with m.when(count == 0):
        reg @= m.uint(1, 32)
    with m.elsewhen(count == 1):
        reg @= m.uint(5, 32)
    with m.elsewhen(count == 2):
        reg @= m.uint(9, 32)
    with m.elsewhen(count == 3):
        reg @= m.uint(13, 32)
    with m.elsewhen(count == 4):
        reg @= m.uint(17, 32)

class Test(m.Circuit):
    io = m.IO(
        CLK=m.In(m.Clock),
        RESETN=m.In(m.ResetN),
        load=m.In(m.Bit),
        O=m.Out(m.UInt[32])
    )
    state = m.Register(m.UInt[len(State)], init=State.IDLE,
                       reset_type=m.ResetN)()
    count = m.Register(m.UInt[clog2(5)], reset_type=m.ResetN)()
    out = m.Register(m.UInt[32], reset_type=m.ResetN)()

    with m.when(state.O == State.IDLE):
        with m.when(io.load):
            state.I @= State.CONFIG
            count.I @= 0
    with m.elsewhen(state.O == State.CONFIG):
        with m.when(count.O < 5):
            config(out.I, count.O)
            count.I @= count.O + 1
        with m.otherwise():
            state.I @= State.IDLE
            count.I @= 0
            out.I @= 0

    io.O @= out.O

rkshthrmsh, 2022-12-21:
@leonardt, based on your recommendation, I can refactor larger chunks of the FSM to make it look cleaner. As a follow up, what about using multiple `return` statements causes the issue, or is it an issue with using `return` at all? I have implemented functions with a single return statement which seemed to be working fine.

leonardt, 2022-12-21:
`return` doesn't work because we cannot overload it in Python, so `return` means simply a Python return statement (it has no hardware semantics).  If you return inside an m.when statement, it will exit the function at that statement (so the subsequent m.when statements won't be executed).  So, basically we can't support the `return` pattern inside m.when without doing some sort of ast rewrite (inline_combinational did this by turning the return statements into effectively the wiring pattern I showed above).

closed 2022-12-22 by rkshthrmsh