phanrahan/magma #1281: pip package not installing: building wheel for coreir fails

rkshthrmsh, 2023-07-06:
It appears that the pip package for magma is not working. I am attempting to install it as per the README: `pip install magma-lang`, but it fails at building wheel for coreir. Installation details and output:
OS: MacOS
Python: Python 3.11
Error output:
```
error: subprocess-exited-with-error

  × Building wheel for coreir (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [22 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.macosx-10.9-universal2-cpython-311
      creating build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/lib.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/util.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/type_gen.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/wireable.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/__init__.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/type.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/generator.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/context.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/simulator.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/global_value.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/module.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/namespace.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      copying coreir/base.py -> build/lib.macosx-10.9-universal2-cpython-311/coreir
      running build_ext
      Cloning into 'coreir-cpp'...
      error: [Errno 2] No such file or directory: 'cmake'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for coreir
Failed to build coreir
ERROR: Could not build wheels for coreir, which is required to install pyproject.toml-based projects
```

leonardt, 2023-07-08:
Working on a change that will make the coreir backend optional

rkshthrmsh, 2023-07-08:
Thank, @leonardt. Is there a workaround for now?

leonardt, 2023-07-11:
Could you try using Python 3.8? Depending on your version of MacOS, you may be able to use the pre existing binary wheel to avoid compiling coreir from source.  Another option is to step through the compile, which will require installing some dependencies like cmake. If you want to go down this path, you can check out the install steps here: https://github.com/rdaly525/coreir/blob/master/INSTALL.md although they are quite old and out of date so you may need to change some things.

leonardt, 2023-07-11:
You can also try the `deprecate-coreir` branch, working on getting the tests updated, but it should hopefully install.

```
git clone https://github.com/phanrahan/magma.git
cd magma
git checkout deprecate-coreir
pip install -e .  # install local symlink package
```

rkshthrmsh, 2023-07-14:
Using `deprecate-coreir` branch worked.

closed 2023-07-14 by rkshthrmsh