phanrahan/magma #1288: [Primitives] `Register(m.Array[N, m.Bit])` doesn't compile

rsetaluri, 2023-07-20:


rsetaluri, 2023-07-20:
```python
class Top(m.Circuit):
    io = m.IO(
        I=m.In(m.Bit),
        O=m.Out(m.Bit),
    ) + m.ClockIO()
    init = None
    # init = 0                                                                                                                           
    # init = m.Array[4, m.Bit]([0, 0, 0, 0])                                                                                             
    reg = m.Register(m.Array[4, m.Bit], init=init)()
    t = m.Array[4, m.Bit]([io.I, io.I, io.I, io.I])
    io.O @= reg(t)[0]
```

Compiling this gives the following error:

```
<stdin>:11:29: error: use of value '%3' expects different type than prior uses: 'i4' vs '!hw.array<4xi1>'
            sv.bpassign %2, %3 : i4
                            ^
<stdin>:9:9: note: prior use here
        %3 = hw.array_create %4, %4, %4, %4 : i1
        ^
```

Looking at the MLIR, the error looks valid

```mlir
module attributes {circt.loweringOptions = "locationInfoStyle=none"} {
    hw.module @Top(%I: i1, %CLK: i1) -> (O: i1) {
        %0 = comb.concat %I, %I, %I, %I : i1, i1, i1, i1
        %2 = sv.reg name "Register_inst0" : !hw.inout<i4>
        sv.alwaysff(posedge %CLK) {
            sv.passign %2, %0 : i4
        }
        %4 = hw.constant 0 : i1
        %3 = hw.array_create %4, %4, %4, %4 : i1
        sv.initial {
            sv.bpassign %2, %3 : i4
        }
        %1 = sv.read_inout %2 : !hw.inout<i4>
        %5 = comb.extract %1 from 0 : (i4) -> i1
        hw.output %5 : i1
    }
}
```

Using `init = m.Array[4, m.Bit]([0, 0, 0, 0])` gives the same issue, while `init = 0` runs into another issue:

```
  File "/Users/rajsekhar/dev/magma-master/magma/generator.py", line 129, in __call__
    this = _make_type(cls, *args, **kwargs)
  File "/Users/rajsekhar/dev/magma-master/magma/generator.py", line 81, in _make_type
    cls.__init__(dummy, *args, **kwargs)
  File "/Users/rajsekhar/dev/magma-master/magma/primitives/register.py", line 181, in __init__
    f"Type {_get_T_from_init(init)} of init ({init}) does not "
ValueError: Type Bits[1] of init (0) does not match T (Array[(4, Bit)])
```

leonardt, 2023-07-20:
For the first issue, perhaps there's somewhere in the code that is checking subclass of `Bits` while somewhere else it's handling array of bit, which might explain i4 instead of array of bit in two different parts of the MLIR

PR #1289 (2023-07-26): [MLIR] Fix array of bit constant creation
Fixes #1288.
commit 8393f222 (2023-07-27): [MLIR] Fix array of bit constant creation (#1289)

Fixes #1288.
commit 41cb1aa0 (2023-08-08): [MLIR] Fix array of bit constant creation (#1289)

Fixes #1288.