phanrahan/magma #1303: pip install -e . fails

phanrahan, 2023-08-09:
I checked out a new version of magma and it did not install.

$ pip install -e .
Obtaining file:///Users/hanrahan/git/magma
  Preparing metadata (setup.py) ... done
Requirement already satisfied: colorlog in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (6.7.0)
Requirement already satisfied: astor in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (0.8.1)
Requirement already satisfied: six in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (1.16.0)
Requirement already satisfied: dataclasses in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (0.6)
Requirement already satisfied: mako in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (1.2.4)
Requirement already satisfied: pyverilog in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (1.3.0)
Requirement already satisfied: numpy in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (1.24.3)
Requirement already satisfied: graphviz in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (0.20.1)
Requirement already satisfied: coreir>=2.0.151 in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (2.0.156)
Requirement already satisfied: hwtypes>=1.4.4 in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages/hwtypes-1.4.7-py3.9.egg (from magma-lang==2.3.0) (1.4.7)
Requirement already satisfied: ast_tools>=0.0.16 in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (0.1.8)
Requirement already satisfied: staticfg in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (0.9.5)
Requirement already satisfied: networkx in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (3.1)
Requirement already satisfied: uinspect in /Users/hanrahan/opt/anaconda3/lib/python3.9/site-packages (from magma-lang==2.3.0) (0.0.5)
INFO: pip is looking at multiple versions of magma-lang to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement circt==1.48.1.dev34 (from magma-lang) (from versions: none)
ERROR: No matching distribution found for circt==1.48.1.dev34

leonardt, 2023-08-09:
Submitted a PR to add support for m-series: https://github.com/llvm/circt/pull/5822

phanrahan, 2023-08-09:
To be clear, the same pip install fails on x86 and m-series.

leonardt, 2023-08-10:
Oh for x86, the issue may be Pythons version, could you try using 3.8? I think circt only has a wheel for for that version

leonardt, 2023-08-10:
It looks like they also have 310, but skipped 39 for some reason, will investigate.

phanrahan, 2023-08-10:
I tried 3.8

ERROR: Ignored the following versions that require a different python version: 0.7 Requires-Python >=3.6, <3.7; 0.8 Requires-Python >=3.6, <3.7; 1.25.0 Requires-Python >=3.9; 1.25.0rc1 Requires-Python >=3.9; 1.25.1 Requires-Python >=3.9; 1.25.2 Requires-Python >=3.9
ERROR: Could not find a version that satisfies the requirement circt==1.48.1.dev34 (from magma-lang) (from versions: none)
ERROR: No matching distribution found for circt==1.48.1.dev34

phanrahan, 2023-08-10:
3.10

ERROR: Ignored the following versions that require a different python version: 0.7 Requires-Python >=3.6, <3.7; 0.8 Requires-Python >=3.6, <3.7
ERROR: Could not find a version that satisfies the requirement circt==1.48.1.dev34 (from magma-lang) (from versions: none)
ERROR: No matching distribution found for circt==1.48.1.dev34

phanrahan, 2023-08-10:
The above was on my m1. It installs on my x86!

phanrahan, 2023-08-10:
pytest test_ice40 works!

phanrahan, 2023-08-17:
I am closing this for now. I do not want to be a PITA.

The install is fragile on m-series macs. coreir doesn't install with current compilers. there are version requirements that are hard to sort out with circt.

closed 2023-08-17 by phanrahan

cdonovick, 2023-08-17:
@phanrahan I believe their is a deprecate-coreir branch which removes the dependency on coreir.  As for circt it is definitely a bit annoying but I have been building from source as I do not use a supported OS.  This admittedly takes many hours on my pretty powerful work station, but is otherwise not particularly difficult.

phanrahan, 2023-08-17:
I agree. The depreciate coreir branch is great. Lenny and Ray can comment in more detail if that is ready to be merged into the main branch. I have noticed their are lots of latent coreir dependencies.

I have all successfully compiled circt from source.  That works for me.