phanrahan/magma #1304: DefineCircuit depreciated

phanrahan, 2023-08-10:
As I port mantle, I am getting warnings that DefineCircuit is depreciated.

Although, I agree we should subclass Circuit and Generator instead of using DefineCircuit. I recommend we keep that low-level interface unless there is a reason not to keep it. I think the builder pattern is worthwhile low-level capability in case the subclassing isn't as convenient.

I think the same could be said for DeclareCircuit.

However, I will convert to the preferred way of creating Circuits.

rsetaluri, 2023-08-10:
Thanks for the feedback. I'm strongly in favor of following the "have one way to do things", unless there's a strong reason to have both. So I think designers should always subclass circuits.

That being said, we can still have a low-level API for creating circuits, perhaps for use internally or testing.

I will add an updated API for this; the old `DeclareCircuit`/`DefineCircuit` were not ideal anyway.

phanrahan, 2023-08-10:
I agree, there should be ideally be a best way to do things. I think subclassing is the best way.

I support updating the low-level methods so their work better.

I think it is ok to have a preferred way and a low-level way. Unless there is no reason to ever need the low-level way.

BTW, when we use repr to print out a Circuit, it uses DefineCircuit....

leonardt, 2023-08-10:
One simple improvement on the define/end circuit could be to use the `with` statement to manage the context

closed 2023-08-16 by phanrahan