phanrahan/magma #1306: DeclareCircuit

phanrahan, 2023-08-12:
One way to implement DeclareCircuit might be as a wrapper around a class creation function.

```
def DeclareCircuit(name, ios=None, params=None):
    return type(name, (Circuit,), {"io":ios, param_types=params))

SB_CARRY = DeclareCircuit("SB_CARRY", IO(I0=In(Bit),  I1=In(Bit),  CI=In(Bit), CO=Out(Bit))
```

leonardt, 2023-08-16:
It's common in Python to use `snake_case` for functions (versus `CamelCase` for objects).  For a "factory" function like this, perhaps we could use a style like `declare_Circuit` to indicate it's a function (`snake_case`) that creates a class (`CamelCase`), so

```python
def declare_Circuit(name: str, io: IO, param_types=None):
    return type(name, (Circuit,), {"io": io, param_types=param_types))

SB_CARRY = DeclareCircuit("SB_CARRY", IO(I0=In(Bit),  I1=In(Bit),  CI=In(Bit), CO=Out(Bit))
```

Also, I added type annotations and made `io` a required parameter since I think magma requires that for a circuit subclass (although perhaps we want to support abstract classes that don't have interfaces?)

phanrahan, 2023-08-16:
As always, your version is better.

I'm closing this since it doesn't need further discussion.

closed 2023-08-16 by phanrahan