phanrahan/magma #1307: Recursive Circuit Definitions

phanrahan, 2023-08-14:
I am trying to port loam to the current version of Magma. Loam has a class Board that may create a class Part each of which is a circuit. This requires recursive circuit definitions.

I get the following err.

```
Exception: Can not instance a circuit outside a definition
```

The old version would allow recursive definitions for exactly this reason. Is there a way to do this in the current version?

rsetaluri, 2023-08-14:
@phanrahan can you provide an example code snippet you're trying to run? It's not immediately clear what recursive circuits have to do with instantiating outside of a circuit context.

For more background, the error message you're seeing is basically disallowing code like

```python
class Foo(m.Circuit): ...

# The next line causes an error. 
foo = Foo()
```

This decision is because an instance outside of a definition really doesn't have much meaning for hardware.

I'm not sure how this relates to recursive circuits.

phanrahan, 2023-08-14:
Let me investigate further.

phanrahan, 2023-08-15:
I believe the inner calls should be subclassing Generator. That would define a Circuit, not an instance. 

What is the difference between Circuit and CircuitType?

Pat

leonardt, 2023-08-16:
The `Circuit` class is a direct subclass of `CircuitType` and adds the `DefineCircuitKind` metaclass.

It seems this is designed to provide alternative mechanisms for creating `CircuitType`s (e.g. without using the metaclass pipeline) that share a common interface with the standard `Circuit` pipeline

phanrahan, 2023-08-16:
Thanks for the clarification.

closed 2023-08-16 by phanrahan

rkshthrmsh, 2023-08-20:
Hi @rsetaluri, I was wondering if something like this would count as 'recursive':

```python3
class Test(m.Circuit):
    io = m.IO(
        en=m.In(m.Bit),
        O=m.Out(m.Bits[4]),
    ) + m.ClockIO()

    count = m.Register(m.Bits[4], has_enable=True)()(
        I = count + 1, # increment by 1 when CE is set
        CE = io.en
    )
    io.O @= count

    # Idiomatic:
    # count = m.Register(m.Bits[4], has_enable=True)()
    # count.I @= count.O + 1
    # count.CE @= io.en
    # io.O @= count.O
```

Of course this does not compile, but it seemed like a concise way of representing a counter instance: 1 statement instead of 3 to instantiate and wire the register IOs.

```
Traceback (most recent call last):
  File "/Users/rkshthrmsh/ws/sandbox/magma_examples/magma_examples/test.py", line 3, in <module>
    class Test(m.Circuit):
  File "/Users/rkshthrmsh/ws/sandbox/magma_examples/magma_examples/test.py", line 10, in Test
    I = count + 1,
        ^^^^^
NameError: name 'count' is not defined. Did you mean: 'round'?
```