phanrahan/magma #1318: How to "group" assign to a Product type?

rkshthrmsh, 2023-09-22:
How to group or bundle assign to a Product type? The following code gives a TypeError.

```python
class Prod(m.Product):
    A = m.Bit
    B = m.Bit

class TestProduct(m.Circuit):
    io = m.IO(
        A=m.In(m.Bit),
        B=m.In(m.Bit),
        O=m.Out(Prod),
    )

    io.O @= Prod[{
        'A': io.A,
        'B': io.B
    }] # TypeError: tuple indices must be integers or slices, not dict

    # Expected:
    # io.O.A @= io.A
    # io.O.B @= io.B
```

leonardt, 2023-09-22:
One way to do this would be an ordered constructor call to `Prod`

```python
class Prod(m.Product):
    A = m.Bit
    B = m.Bit

class TestProduct(m.Circuit):
    io = m.IO(
        A=m.In(m.Bit),
        B=m.In(m.Bit),
        O=m.Out(Prod),
    )

    io.O @= Prod(io.A, io.B)
```

It doesn't look like we support named key/value pairs for the constructor for Product types, which does seem like something we should.  For now, if you would like to use key/value pairs, you could use the `product` conversion helper function:

```python
class Prod(m.Product):
    A = m.Bit
    B = m.Bit

class TestProduct(m.Circuit):
    io = m.IO(
        A=m.In(m.Bit),
        B=m.In(m.Bit),
        O=m.Out(Prod),
    )

    io.O @= m.product(A=io.A, B=io.B)
```

In this case, magma will create an anonymous product type that will be structurally compatible with `Prod` since they have the same keys.

rsetaluri, 2023-09-22:
Using `m.product()` helper would be the most recommended way to do this.

rkshthrmsh, 2023-09-23:
Thank you! Both options work well. The way I see it, the `Prod(io.A, io.B)` could be used to give more context for a reader when the code uses multiple `Product` types, whereas `m.product(A=io.A, B=io.B)` is more generic. 

Closing issue since it's tracked in 1319.

closed 2023-09-23 by rkshthrmsh