phanrahan/magma #1327: Creating intermediate values using `Wire` gives `TypeError`

rkshthrmsh, 2023-09-24:
In the following code, I am trying to create a (verilog) `wire` to store an intermediate value. However, this gives a `TypeError: unsupported operand type(s) for @=: 'Wire' and 'SInt[(2, Out(Bit))]'`. 

```python
class Test1(m.Circuit):
    io = m.IO(
        A=m.In(m.SInt[2]),
        En=m.In(m.Bit),
        O=m.Out(m.SInt[2]),
    )
    temp = m.Wire(m.SInt[2])
    with m.when(io.En):
        temp @= io.A
    with m.otherwise():
        temp @= 0    
    
    io.O @= temp + 1
```

The equivalent verilog would be
```verilog
assign temp = En? A : 0;
```
How can this be done?

rkshthrmsh, 2023-09-24:
Figured out there is no need to declare a `Wire`. Instead, calling the data type `SInt` serves the same purpose. 
```python
    temp = m.SInt[2]()
    with m.when(io.En):
        temp @= io.A
    with m.otherwise():
        temp @= 0
```

closed 2023-09-24 by rkshthrmsh

rsetaluri, 2023-09-25:
Yes @rkshthrmsh good observation. To follow up on that you should use `temp = T()` as you do (with `T = m.Sint[2]` in your case, for example). You *can* use the `m.Wire()` primitive, but it is a circuit like any other circuit, so you would have to do:

```python
temp_wire = m.Wire(m.SInt[2])()  # note the extra pair of paren's
with m.when(...):
    temp_wire.I @= ...  # note we do .I to get the input of this primitive
with m.otherwise():
    temp_wire.I @= ...
io.O @= temp_wire.O + 1
```