phanrahan/magma #1329: Using input types to index Array register gives TypeError

rkshthrmsh, 2023-09-25:
How to use an input to index into an array register? The following is a minimal example, along with the TypeError it generates.

```python
class Test1(m.Circuit):
    io = m.IO(
        A=m.In(m.UInt[2]),
        B=m.In(m.UInt[4]),
        O=m.Out(m.UInt[4]),
    )
    arr = m.Register(m.Array[4, m.UInt[4]])()
    arr.I[io.A] @= io.B
    io.O @= arr.O[0]
```
```
TypeError: mux arg I[0] (.I[0]: In(UInt[4])) does not match inferred input port type UInt[4]
```

leonardt, 2023-09-25:
Is the idea here to dynamically set one index of `arr`? If so, you could try using `set_index` https://github.com/phanrahan/magma/blob/e0a49bb3079a9938526748356fb9556fea2962d0/magma/primitives/set_index.py#L7-L13

```python
class Test1(m.Circuit):
    io = m.IO(
        A=m.In(m.UInt[2]),
        B=m.In(m.UInt[4]),
        O=m.Out(m.UInt[4]),
    )
    arr = m.Register(m.Array[4, m.UInt[4]])()
    arr.I @= m.set_index(arr.O, io.B, io.A)
    io.O @= arr.O[0]
```

I'll have to investigate this error which is not very helpful, but I don't think this syntax would necessarily work in general because you'd want to specify a default value for the values that aren't written.  The function `set_index` style implies that the value is unchanged for the indices which I think is more explicit.

rkshthrmsh, 2023-09-26:
Hi @leonardt, yes, the intention was to dynamically set specific indices of a register array. As you mentioned, in the `arr.I[io.A] @= io.B` syntax, not specifying a default value for other indices is the issue. 

In a `Register`, could the previous `O` values be considered as the default for `I`s that are not explicitly wired? Are there potential issues with that?

rkshthrmsh, 2023-09-26:
As an extension, how can one dynamically set an index of an N-D array?
Based on the function signature of `set_index`, it appears that `idx` is `UInt` only.

```python3
class Test1(m.Circuit):
    io = m.IO(
        X=m.In(m.UInt[2]),
        Y=m.In(m.UInt[2]),
        B=m.In(m.UInt[4]),
        O=m.Out(m.UInt[4]),
    )
    arr = m.Register(m.Array[(4, 4), m.UInt[4]])()
    # Intention:
    # arr.I[io.Y][io.X] @= io.B
    arr.I @= m.set_index(arr.O, io.B, (io.X, io.Y)) # TypeError: Expected value to be the same type as `target`'s contents
    io.O @= arr.O[0][0]
```

rkshthrmsh, 2023-09-26:
Figured out this can be done recursively like so:
```python
    arr.I @= m.set_index(arr.O, m.set_index(arr.O[io.Y], io.B, io.X), io.Y)
    # Equivalent to:
    # arr.I[io.Y][io.X] @= io.B and all other inputs wired to defaults
```
However, the syntax can look messy when dealing with N-D arrays having multiple indices. Thought it might be better to simplify it to:
```python
    arr.I @= m.set_index(arr.O, io.B, [io.Y, io.X])
```

Have opened a PR for this.

rkshthrmsh, 2024-03-04:
Merged in #1353

closed 2024-03-04 by rkshthrmsh