phanrahan/magma #1358: Installation on latest `main` branch is failing

rkshthrmsh, 2024-03-04:
Related to #1303 
Installation on the latest `main` branch fails with the following error on `Python 3.9` (and `Python 3.10`). 

```
pip3.9 install -e .
Obtaining file:///Users/rkshthrmsh/ws/magma
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Installing backend dependencies ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting colorlog (from magma-lang==3.0.1)
  Using cached colorlog-6.8.2-py3-none-any.whl.metadata (10 kB)
Collecting astor (from magma-lang==3.0.1)
  Using cached astor-0.8.1-py2.py3-none-any.whl.metadata (4.2 kB)
Collecting six (from magma-lang==3.0.1)
  Using cached six-1.16.0-py2.py3-none-any.whl.metadata (1.8 kB)
Collecting dataclasses (from magma-lang==3.0.1)
  Using cached dataclasses-0.6-py3-none-any.whl.metadata (3.0 kB)
Collecting mako (from magma-lang==3.0.1)
  Using cached Mako-1.3.2-py3-none-any.whl.metadata (2.9 kB)
Collecting pyverilog (from magma-lang==3.0.1)
  Using cached pyverilog-1.3.0.tar.gz (136 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Collecting numpy (from magma-lang==3.0.1)
  Downloading numpy-1.26.4-cp39-cp39-macosx_11_0_arm64.whl.metadata (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.1/61.1 kB 781.6 kB/s eta 0:00:00
Collecting graphviz (from magma-lang==3.0.1)
  Using cached graphviz-0.20.1-py3-none-any.whl.metadata (12 kB)
Collecting coreir>=2.0.151 (from magma-lang==3.0.1)
  Using cached coreir-2.0.156.tar.gz (18 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Collecting hwtypes>=1.4.4 (from magma-lang==3.0.1)
  Using cached hwtypes-1.4.7.tar.gz (30 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Collecting ast-tools>=0.0.16 (from magma-lang==3.0.1)
  Using cached ast_tools-0.1.8-py3-none-any.whl.metadata (4.8 kB)
Collecting staticfg (from magma-lang==3.0.1)
  Using cached staticfg-0.9.5-py3-none-any.whl.metadata (313 bytes)
Collecting networkx (from magma-lang==3.0.1)
  Using cached networkx-3.2.1-py3-none-any.whl.metadata (5.2 kB)
Collecting uinspect (from magma-lang==3.0.1)
  Downloading uinspect-0.0.5-cp39-cp39-macosx_10_15_universal2.whl.metadata (2.8 kB)
INFO: pip is looking at multiple versions of magma-lang to determine which version is compatible with other requirements. This could take a while.
ERROR: Ignored the following versions that require a different python version: 0.7 Requires-Python >=3.6, <3.7; 0.8 Requires-Python >=3.6, <3.7
ERROR: Could not find a version that satisfies the requirement circt==1.59.1.dev51 (from magma-lang) (from versions: none)
ERROR: No matching distribution found for circt==1.59.1.dev51
```

rkshthrmsh, 2024-03-04:
Actually, it's also failing on the latest `deprecate-coreir` branch:

```
pip3.9 install -e .
DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
Obtaining file:///Users/rkshthrmsh/ws/magma
  Preparing metadata (setup.py) ... done
Collecting colorlog (from magma-lang==2.3.0)
  Using cached colorlog-6.8.2-py3-none-any.whl.metadata (10 kB)
Collecting astor (from magma-lang==2.3.0)
  Using cached astor-0.8.1-py2.py3-none-any.whl.metadata (4.2 kB)
Collecting six (from magma-lang==2.3.0)
  Using cached six-1.16.0-py2.py3-none-any.whl.metadata (1.8 kB)
Collecting dataclasses (from magma-lang==2.3.0)
  Using cached dataclasses-0.6-py3-none-any.whl.metadata (3.0 kB)
Collecting mako (from magma-lang==2.3.0)
  Using cached Mako-1.3.2-py3-none-any.whl.metadata (2.9 kB)
Collecting pyverilog (from magma-lang==2.3.0)
  Using cached pyverilog-1.3.0.tar.gz (136 kB)
  Preparing metadata (setup.py) ... done
Collecting numpy (from magma-lang==2.3.0)
  Using cached numpy-1.26.4-cp39-cp39-macosx_11_0_arm64.whl.metadata (61 kB)
Collecting graphviz (from magma-lang==2.3.0)
  Using cached graphviz-0.20.1-py3-none-any.whl.metadata (12 kB)
Collecting hwtypes>=1.4.4 (from magma-lang==2.3.0)
  Using cached hwtypes-1.4.7.tar.gz (30 kB)
  Preparing metadata (setup.py) ... done
Collecting ast_tools>=0.0.16 (from magma-lang==2.3.0)
  Using cached ast_tools-0.1.8-py3-none-any.whl.metadata (4.8 kB)
Collecting staticfg (from magma-lang==2.3.0)
  Using cached staticfg-0.9.5-py3-none-any.whl.metadata (313 bytes)
Collecting networkx (from magma-lang==2.3.0)
  Using cached networkx-3.2.1-py3-none-any.whl.metadata (5.2 kB)
Collecting uinspect (from magma-lang==2.3.0)
  Using cached uinspect-0.0.5-cp39-cp39-macosx_10_15_universal2.whl.metadata (2.8 kB)
INFO: pip is looking at multiple versions of magma-lang to determine which version is compatible with other requirements. This could take a while.
ERROR: Ignored the following versions that require a different python version: 0.7 Requires-Python >=3.6, <3.7; 0.8 Requires-Python >=3.6, <3.7
ERROR: Could not find a version that satisfies the requirement circt==1.48.1.dev34 (from magma-lang) (from versions: none)
ERROR: No matching distribution found for circt==1.48.1.dev34
```

leonardt, 2024-03-04:
What operating system are you using? I think we may need to try updating the circt dependency and they dropped wheels for python versions except 3.8 and 3.10, but those older ones should be compatible. Are you on an M series Mac? I think there might not be a wheel available for that yet.

leonardt, 2024-03-04:
As a temporary workaround, you could try building the Python package from source using these instructions: https://github.com/llvm/circt/blob/main/docs/PythonBindings.md

rkshthrmsh, 2024-03-05:
Yes, I am on an M series Mac.

leonardt, 2024-03-05:
There's currently some issue with deploying an m-series wheel (see https://github.com/llvm/circt/issues/6153), until that's resolved you'll need to build from source